Ultimaker/Cura #7807: [4.6.1] First run stuck at disclaimer

matthieu-vergne, 2020-05-23:
**Application version**
4.6.1

**Platform**
Kubuntu 20.04

**Printer**
NA

**Reproduction steps**
Run AppImage after download.

**Screenshot(s)**
[image]

**Actual results**
Getting started then disclaimer pages show, but the page does not fit the screen so I cannot access the button to go further.

**Expected results**
The page should fit the screen so we can access all the text and buttons.

**Project file**
NA

**Log file**
NA

**Additional information**
NA

jellespijker, 2020-05-23:
Hi Matthieu,

Are you running the AppImage or did you install it from the snapstore?

It might also help us if you share your log files.

matthieu-vergne, 2020-05-23:
Just added the log file and clarified the steps to highlight it is the AppImage.

rburema, 2020-05-24:
What is your screen resolution?

matthieu-vergne, 2020-05-24:
1920x1080

Ghostkeeper, 2020-05-26:
Hmm, yeah I think the problem is that Cura enforces a minimum size on the window, but that is a fixed number of pixels. It doesn't take the pixel scaling factor into account then.

matthieu-vergne, 2020-05-26:
Is there a chance for it to be fixed soon?
Or is there a workaround?
So far, I see no button at all except "Decline and close" if I turn the screen.
So I just cannot use it since I cannot go further than this page.

jellespijker, 2020-05-26:
I can reproduced this scenario on Manjaro with KDE, by setting the scaling to 200% (in my case).

A work around might be to go to the `display Configuration - System Settings` and change the global scale to 100% assuming that you are now using an scale setting which is higher

matthieu-vergne, 2020-05-26:
The global scale is at 100%. However, the fonts have been increased for a better reading (around 50%, e.g. 10pt -> 14pt). Decreasing the general font size works. Moreover, it does not need to reconnect like the global scale setting, so it is easier as a workaround.

However, it may be a hint that Cura is not adapted to computers with accessibility settings enabled, since they may increase the size of the fonts among other things. I didn't test it though. So maybe the issue should be investigated as a more general issue about accessibility.

jellespijker, 2020-05-26:
We understood the hint and we discussed today how we could make it more user friendly in that regard.

matthieu-vergne, 2020-05-26:
{^_^}b

From here, I consider that the provided workarounds are relevant solutions to the specific problem of this issue, and that a proper fix should be considered through a broader accessibility perspective, which is out of the scope of this issue. So let's close this one and hope you the best to find a nice way to improve user friendliness.

Thank you all! {^_^}°

closed 2020-05-26 by matthieu-vergne

Ghostkeeper, 2020-05-27:
Indeed, all of the scale in Cura is tied to the font size. We had trouble getting the layout to work correctly otherwise.

Except the minimum window size, apparently.