Ultimaker/Cura #7808: 4.6.1 support interface inconsistency

batesman1, 2020-05-23:
**Application version**
4.6.1

**Platform**
Win 10

**Printer**
Ender 3 Pro

**Reproduction steps**
Added model and sliced with standard supports and support interface enabled. Additionally there are odd holes in the support that shouldn't be there. If I switch to tree supports, the interface is present.

**Screenshot(s)**
[image]

[image]

**Actual results**
Slicer fails to add support interface when it should.

**Expected results**
The support roof and floor should be present where the support touches the model and the base.

**Project file**
[CE3PRO_Cultist_Body_NEW.zip](https://github.com/Ultimaker/Cura/files/4671208/CE3PRO_Cultist_Body_NEW.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4671212/cura.log)

**Additional information**
(Extra information relevant to the issue.)

64Florian, 2020-05-24:
Hello @batesman1 
In my point of view the top and bottom need at least one "gap" to let the user to be able to remove the support without braking the product ..
-> **Your product is very small (height less than 4 cm)**  
In you case, I confirm for your bottom of the support, it's a bit strange:
[image]

But if I revert your setting (by selecting again "Super quality" -> "Discard" ) 
then enable support and line I have a "good" support: 
[image]
**Please ! before doing that, save your profile**, otherwise you'll lose all your settings (_ok you have a backup on this forum now in the 3mf file .._ )

For dev and support teams: discard "clean" following settings:
[image]
[image]

Sorry, for the moment, I was not able to find "why" the bottom of the support doesn't start where it should...

rburema, 2020-05-24:
@batesman1  Could you try to setting either (or both) 'Support Stair Step Width' or 'Support Stair Step aximum Height' to 0? Just trying to confirm if part of this is related to a known issue (that we maybe finally getting to soon).

Ghostkeeper, 2020-05-26:
Reduce the Minimum Support Area to 0 to make the support re-appear.

Reduce the Minimum Support Interface Area to 0 to make the support interface re-appear.

Those settings seem to be working as expected, so I'll close this issue.

closed 2020-05-26 by Ghostkeeper

batesman1, 2020-05-27:
Confirmed. Resetting the minimum support interface area did the trick. Thanks to you all!