Ultimaker/Cura #7809: Cura 4.6.1 prints outside the Printbed - Anycubig Mega.

Lightboxx, 2020-05-23:
**Application version**
Version 4.6.1

**Platform**
Windows 10 Build 1903 - Ryzen 3700x 64GB Ram

**Printer**
Anycubic Mega i3

**Reproduction steps**
the prints are not centered on the printbed. 
it´s off like shown in the attached topview image (~1cm and ~0.5cm).
when i wanted to print the big model the brim was printed outside the printbed.

when i home all axis the print nozzle is exactly at the lower left corner of the printbed.
so i don´t understand why the prints with cura are off.

also the printhead settings in cura only allow negativ numbers. 
i have no idea if this has something to do with my print problem.
but i was told to input positive numbers, but that does not work in cura v4.6.1.
i did not change these settings.
only negative numbers are accepted by cura.

**Screenshot(s)**
[image]
[image]
[image] 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

64Florian, 2020-05-24:
Hello @Lightboxx
> i removed cura 4.6.1 and all remainings.

Did you try to "cleaup" your profile too ? 
Just in case, 
- open Cura, 
 - click on "Help" -> "Show configuration folder" -> you should have a new file explorer opened in a 4.6 folder ...
 - Close Cura
 - rename the 4.6 folder to 4.6_Ko 
 - restart Cura 

Thank you

64Florian, 2020-05-24:
> [image]

in fact, red circle are for the "size of the head" with the nozzle as reference (those settings are used when you define _print sequence_ on "one at the time" to prevent collision of the head and a previously printed object)

Lightboxx, 2020-05-25:
why are only negative numbers allowed? 
people tell me i should use positive numbers. but cura won´t accept positive numbers as i wrote.

> Did you try to "cleaup" your profile too ?

yes as i wrote, i removed cura 4.6.1 and all remainings.
i was even ging through the windows registry.

64Florian, 2020-05-25:
Hello @Lightboxx 
In fact registry and 4.6.1 folder are **Not** your profiles folder ... 
Please try to do this : 

  -  open Cura,
   - click on "Help" -> "Show configuration folder" -> you should have a new file explorer opened in a 4.6 folder ...
   - Close Cura
   - rename the 4.6 folder to 4.6_Ko
   - restart Cura

gravesfamily, 2020-05-26:
I have the same issue, I think.

New install of Win10, so I had to install a fresh copy of Cura 4.6.1. I had written down a bunch of my settings, including the printhead settings, the ones I've been using successfully for years. Those settings are:
X min 35mm
Y min 70mm
X max 55mm
Y max 60mm

However, using the dialog box, I cannot make the xmin or ymin anything other than a negative number. Fortunately, I am able to edit the inst.cfg file in the definition_changes folder and set the line to be:
machine_head_with_fans_polygon = [[35, 60], [35, 70], [55, 60], [55, 70]]

The awesome thing is that if I open Cura and go into the dialog box, it still shows 0 for the xmin and ymin fields. Fortunately, it doesn't seem to rewrite those numbers if I edit something else, like the start gcode.

I did try running Cura as an admin, but that didn't change anything. I also went back to Cura 4.3 and it had a similar issue, so this has been around for awhile, at least in my Windows configuration.

Ghostkeeper, 2020-05-26:
> people (some) always tell me i should put in positive numbers to fix this problem.

That information used to be correct, but is outdated now. These are the coordinates of the edges of the print head, relative to the nozzle position. Assuming your nozzle is inside your print head, the two "min" values must be negative and the two "max" values must be positive.

> others say the printhead settings have no influence except when i choose to print "one at a time" in cura.

That's correct. So let's not talk about these because they really don't have anything to do with your issue.

Real problem
----
If the build volume is not centred on the physical build plate, then either the printer dimensions are wrong, the endstops are wrong on the printer, or the firmware is wrong.

The printer dimensions are [listed](https://www.anycubic.com/products/anycubic-i3-mega) as 210x210x205, which is also the default in Cura so that part seems to be correct to me.

The endstops determine on your printer where the 0,0 coordinate is. So you can try adjusting those to centre the print better.

Sometimes the firmware applies an offset in an attempt to centre the print. I don't expect that to be wrong if Anycubic did that themselves, but maybe if you loaded up your custom firmware you need to adjust that offset as well.

Lightboxx, 2020-05-26:
hi,

thx for the answers. 

the endstop for the y-axis (which is most affected) is fixed on my printer. i can´t adjust that.
and when i home all axis the printhead moves to the lower left edge of the printbed , that seems to be ok too.

how can i check if the gcode is centered to the bed? 
i have no clue about gcode commands.

Ghostkeeper, 2020-05-27:
To check if anything is centred to the bed, you have to measure with a ruler. The printer has no sensor of where the bed is, so you can't tell by just the g-code or the display.

If you open the g-code file with a text editor you'll find loads of `G0` and `G1` commands. Those tell your printer to move somewhere, and they usually have coordinates with them to tell the printer where it needs to move. I'd expect that those correspond with the coordinates you see on that display.

GregValiant, 2024-12-15:
It's been years so I'll go ahead and close this.  Let me know if that is premature.  The problem I see here is the "Home Offset" setting in the printer so it is not a Cura issue.

closed 2024-12-15 by GregValiant