Ultimaker/Cura #7810: [4.6.1] support does not touch buildplate

berberic2, 2020-05-23:
**Application version**
4.6.1

**Platform**
Linux

**Printer**
Monoprice Delta Mini

**Problem**
If _Support Placement_ is set to _Everywhere_ Cura make a gap below support that is based on the buildplate and does not startr al first layer. This makes support free floating and (small) support structures fail.
The gap is necessary if the support sits on top a build part, but not on support that sits on the buildplate.

**Resolution**
Support that is based on the buildplate should **always** start at first layer without a gap, regardless of settings.

berberic2, 2020-05-23:
Screenshots of layer 1, 2, 3 and 4, showing that support in the holes start at layer **4**.

[image]
[image]
[image]
[image]

64Florian, 2020-05-23:
Hello,
Could you add an example with your CURA with your settings and click on "File" -> "Save" ->"Save" -> "Save" (after providing a filename)
The output file is an "3mf" file. Please Zip the file and drop it in a post in your ticket.
=> This file is really useful to see / understand / reproduce the same behavior as you describe
(_As I know / see I'm not able to reproduce on my version_)

Thank you

64Florian, 2020-05-24:
For dev team / as memo: could it be the same issue as #7808

berberic2, 2020-05-24:
sure.
Project file of example above.
[MMD_Part_1.3mf.zip](https://github.com/Ultimaker/Cura/files/4673477/MMD_Part_1.3mf.zip)

64Florian, 2020-05-24:
Thank you @berberic2 
unfortunately, It seems something missing in the file to make the same workspace as yours 
[image]
I'll try to use normal default but with my settings I'm not able to reproduce 
[image]

Could you go in Cura, -> "Help" -> "Show configuration folder" -> a file explorer should be opened with a 4.6 folder selected 
Pleas zip this folder and add it in this ticket (_this folder contain all your printer profile_)

64Florian, 2020-05-24:
Just for you,
For this product, you should use "Grid" @ 0% infill... your product should be better printed (_In my point of view_)
but it's another thing

berberic2, 2020-05-24:
The .3mf contains the machine definition and the filament definition. 
```
.
├── 3D
│   └── 3dmodel.model
├── [Content_Types].xml
├── Cura
│   ├── 3djake_pla_silber.xml.fdm_material
│   ├── monoprice_mini_delta.def.json
│   ├── monoprice_mini_delta_extruder_0 #2.extruder.cfg
│   ├── monoprice_mini_delta_extruder_0 #2_settings.inst.cfg
│   ├── monoprice_mini_delta_extruder_0 #2_user.inst.cfg
│   ├── monoprice_mini_delta_extruder_0.def.json
│   ├── Monoprice Mini Delta.global.cfg
│   ├── monoprice_mini_delta_Global_Fast_Quality.inst.cfg
│   ├── Monoprice Mini Delta_settings.inst.cfg
│   ├── Monoprice Mini Delta_user.inst.cfg
│   ├── preferences.cfg
│   └── version.ini
├── _rels
└── X.zip
```

So this seems to be another bug in cura, in the project load or save routine.
_Grid_ for _**support-**_ or _**infill-**_pattern and 0 % _**infill**_ ?!

rburema, 2020-05-24:
@berberic2 Could you try to setting either (or both) 'Support Stair Step Width' or 'Support Stair Step aximum Height' to 0? Just trying to confirm if this is a known issue (that we maybe finally getting to soon).

berberic2, 2020-05-25:
> Could you try to setting either (or both) 'Support Stair Step Width' or 'Support Stair Step aximum Height' to 0?

Setting either of these values to 0 solves the problem.

Ghostkeeper, 2020-05-26:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/6379.

closed 2020-05-26 by Ghostkeeper