Ultimaker/Cura #7811: Concentric bottom pattern initial layer prints in sub-optimal order

sigmatechnica, 2020-05-23:
The Concentric option on the "bottom pattern initial layer"  prints inside to outside.  When printing  with stringy sticky filaments like PET this is undesirable since a) stringing occurs on the travel to the middle of the first layer to start, and this can spoil the appearance of the first layer. b) on some print beds (i'm using printbyte) the very small starting patch does not adhere properly to the build plate and gets dragged around by the print head, causing a lump/hole/messed up patch in the middle of the layer.  

To fix this a sub-option to the concentric bottom pattern initial layer should be added to print from the outside towards the inside.

smartavionics, 2020-05-25:
If you wish to try my latest build (20200525) which should be on dropbox in the next hour it has a fix for this issue.

You can find my builds at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0&lst=

Please read the README.md file there before using.

Ghostkeeper, 2020-05-26:
Did you try printing from the outside to the inside? Our experiences with this order have been terrible. It tends to produce a massive blob on the inside since any overextrusion gets pushed inwards and is trapped there. This is especially true for the first layer which is prone to have overextrusion.

sigmatechnica, 2020-06-05:
smartavionics: Thanks! i'll give that a go tomorrow!
Ghostkeeper: I can see how that might be a problem.  Will try smartavionics build and report back

sigmatechnica, 2020-06-06:
I tested build 20200525 today and it worked perfectly.  No blobbing, just a perfect concentric bottom layer.  Admittedly it was a small part, about 15mm so less opportunity for the accumulation of excess plastic but it did leave a very nice finish.   It would be a worth while addition as an optional feature IMO, people can adjust over extrusion, not so much they can do about filament just not sticking at the start of a concentric layer printed from the middle

mahtDFR, 2020-06-12:
I've added this to our backlog as CURA-7499

fvrmr, 2021-06-01:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features from our community and from Ultimaker, we have decided to remove this feature request from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

Leapo, 2023-07-08:
I'm still seeing this same issue on Cura 5.3 and 5.4, even though the release notes for Cura 5.3 explicitly state that it was fixed:

> Now, when selecting the ‘Concentric ‘ option under the ‘Bottom Pattern Initial Layer’ setting, your first layer will print sequentially from the outside in.

The "concentric" pattern for the initial layer should print outside-to-in, but it's still printing inside-to-out for me.

Leapo, 2023-07-20:
An update on this? The Cura 5.3 release notes and announcement both say this was fixed:
https://ultimaker.com/learn/unlock-true-multi-material-printing-with-ultimaker-cura-5-3/

The above even includes an animated GIF, showing the correct outside-to-in behavior... but 5.3 and 5.4 both still print inside-to-out when I try them.