Ultimaker/Cura #7812: Support Octoprint Cancel Object Plugin

Sophist-UK, 2020-05-23:
**Is your feature request related to a problem? Please describe.**
AFAIK at present Octoprint Cancel Object plugin cannot work with Cura Gcode because it doesn't contain comment markers for each layer to show when the print for one object / object-part has completed and another is beginning.

**Describe the solution you'd like**
Provide a post-processing plugin - or more likely Cura-Engine base functionality - to add the markers that Octoprint Cancel Object needs to be able to identify the switch points in the Gcode and skip the objects that have been cancelled by the user part way through the print.

**Describe alternatives you've considered**
None.

64Florian, 2020-05-23:
Hello,
Do you know how the works ?
Because, as I can see I have few comment in my GCode 
;LAYER:1
M140 S60
M106 S255
;TYPE:WALL-OUTER
;MESH:Cube 1cm B.stl
....
;MESH:Cube 1cm.stl
...
;TIME_ELAPSED:46.347394
;LAYER:2
;TYPE:WALL-OUTER
;MESH:Cube 1cm B.stl
...
;TYPE:WALL-INNER
...
;TYPE:SKIN
...
;MESH:Cube 1cm.stl
...
;TYPE:WALL-OUTER
...
;TYPE:WALL-INNER

[image]

Sophist-UK, 2020-05-23:
I was perhaps a bit premature on raising this. Having now checked the [Octoprint Cancel Object Github Page](https://github.com/paukstelis/OctoPrint-Cancelobject) I find that the readme.md says:
```
Cura (1.3.6 or greater)
Objects are automatically tagged with the ;MESH:<stl file name> tag.

Please note that the way that Cura does supports means that ALL supports are placed into an object called NONMESH. Cancelling an object will not cancel its supports.

Do not use stl files that have non-ASCII characters. This will confuse all current versions of Marlin.
Is is a good idea to add ;MESH:ENDGCODE at the start of your ending gcode script (no spaces after ENDGCODE!)
```

So, a minor enhancement to supports so that they are tagged with (say) `;MESH:objectname`  `and ;TYPE:SUPPORT` instead would be both a better description and allow supports to be cancelled with the base object. Custom supports may require something different as (I have a feeling that) these are handled inside Cura as additional objects.

Ghostkeeper, 2020-05-26:
The non-ASCII characters thing has been fixed in the meanwhile. It replaces them with a question mark: https://github.com/Ultimaker/CuraEngine/blob/39bb08c238359d8f1a1fbf7f775b31fc0cc2598a/src/gcodeExport.cpp#L24-L34

Ghostkeeper, 2020-05-26:
Is there anything specific that we still need to do? As far as I can tell, the plug-in has grown to support us, rather than the other way around.

shatteredsite, 2020-06-07:
The ability to cancel the supports per-object is kind of essential for this to function properly with Cura gcode. If I misread and supports are indeed separate objects associated with the object they support, please ignore.

thattolleyguy, 2021-01-05:
@Ghostkeeper Can you explain why all supports need to be part of the same MESH? Wouldn't it make more sense to have them as part of each objects mesh but a separate TYPE? I don't understand how this would affect dual extrusion. 

As far as a plug-in not able to do it, it does it fine for other slicers. I'd love to be able to get cura to support this.

Ghostkeeper, 2021-01-18:
It's not necessarily a "mesh" for support. Basically the support are all generated at once for the entire print.

There are several good reasons to implement it like this. The most common one being dual extrusion. To illustrate, let's say you're printing a model like this:
[image]

This is actually two models that happen to mesh perfectly together. If the support were two different pieces beneath them, you'd see the same spotted pattern in the support underneath the two models through broken-up lines, extra zig-zags for connected support infill lines, and extra walls in the middle if Support Wall Line Count is >0. It would print first the support underneath one of the models and then underneath the other. This would indeed be necessary if you want to be able to cancel one of the two. But if you don't, then that is disastrous for the quality and reliability of the support. You want it to print the support with continuous lines, not interrupting them to print part of the line later, and not with extra lines in the middle.

Aside from that, we also have the settings [Support Join Distance](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/support/support_join_distance.md), [Conical Support](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/experimental/support_conical_enabled.md) and [Tree Support](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/support/support_structure.md#tree) which wouldn't be able to join support pieces together if they need to be kept separate.

thattolleyguy, 2021-01-18:
@Ghostkeeper So how do other slicers handle this? Would it be possible to provide a setting that allows you to change between object specific supports and one support? 

What percentage of cura users have multiple extruders? I understand wanting to keep the currently supported feature set but it doesn't seem like you couldn't support both.

Ghostkeeper, 2021-01-26:
I don't have experience with multi-extrusion in Simplify3D. It's been a while since I used that application.

I know that Slic3r just labels the whole of support as belonging to one of the objects. So then if you cancel the wrong object it would cause the other objects to suddenly be unsupported, unless this plug-in you're using has an explicit exception to not cancel supports built-in (then it would behave like with Cura's g-code).

Ghostkeeper, 2021-01-26:
> What percentage of cura users have multiple extruders?

Somewhere around 15% I'd guess. It's out of the question that we'd diminish the experience of dual-extrusion prints though since it's one of Ultimaker's biggest selling points.

To be able to mark different parts of support as belonging to different objects, we'd need to keep the support completely separate per object. This necessarily splits up the support area and will therefore create borders between the different support areas. In the best case it would print small segments of support at a time, clustering parts beneath one object together. In the worst case, the user has extra support walls or connected support lines enabled and would get much too strong support that can't be removed from the print any more.

One way to implement it better would be to first generate the complete support and then cut that up without applying the extra infill lines or connected infill. That is then merely a problem for the buffer underrun (introducing extra commands that are smaller than the Maximum Resolution setting). However that doesn't solve the problem for Conical Support or Tree Support where a branch may be supporting two different objects and cancelling either of them should not remove the branch. And it's also basically going around our current architecture so it'd need refactoring.