Ultimaker/Cura #7814: Cura cannot slice

hrhlasvegas, 2020-05-24:
I use Cura latest version on Mac OS Catalina. All of the sudden Cura doesn't slice anymore. I click slice and it stays stuck.

64Florian, 2020-05-24:
Hello @hrhlasvegas 
Could you add a screenshoot please ?
-> Are you sure your product "fit" on your build plate ?
[image]

Can you "export" your workspace: "File" -> "Save" -> "save" -> enter a filename -> "save"
Now zip the file and add it in this ticket please.

Thank you

hrhlasvegas, 2020-05-24:
There you go. You can see you slicer starts and gets stuck there.

Marco

64Florian, 2020-05-24:
sorry, I was not able to see anything in the ticket :-S

hrhlasvegas, 2020-05-24:
Try now

[reultimakercuracuracannotslice7814.zip](https://github.com/Ultimaker/Cura/files/4674304/reultimakercuracuracannotslice7814.zip)

64Florian, 2020-05-24:
This time I'm able to see and download your file, Thank you !

Unfortunately, I'm not able to reproduce your issue (but I'm not on the same version of MacOS as yours) 

are you allays stuck like that or is it only randomly ?

rburema, 2020-05-24:
@hrhlasvegas  Can you please fill out the template? One of the things it asks for are the _logs_ which may prove to be quite useful diagnosing connections to the slicing engine for example.

hrhlasvegas, 2020-05-24:
I uninstalled Cura and all the settings folders and started over. Now it works... really weird

rburema, 2020-05-24:
The previous installation may have been corrupted somehow. The unresponsive engine is also a frequent tell in such cases.

Asterchades, 2020-05-26:
Sometimes you can "jiggle the handle" a bit, so to speak, by repositioning one of the models on the print bed (you can immediately put it back where it came from). I see this occur often if you hit the "Slice" button too quickly after having changed a print setting (though it possibly applies to other things as well). By "too quickly" I mean only a faction of a second, or around the time it takes for the previous Slice preview to be invalidated (I suspect it could be related to this).

closed 2020-06-19 by no-response[bot]