Ultimaker/Cura #7815: Question: Using Cura for 5axes 3dprinter

l1va, 2020-05-24:
Hi, 
Is it difficult to change sources or by plugins or with something else to add support  5axes 3d printing. I mean rotating and tilting platform (bed). Is it possible to slice not by Z? 
If yes, point please where it is better to start, what code should be updated firstly. 
By fast review seems it is not so easy - all algorithms use such base that slicing is always by Z.

64Florian, 2020-05-24:
Hello,
Do you means "_3D Printer Tilted Conveyor Bed_" approach ? 
https://www.youtube.com/watch?v=ynCYqWbT4t0

In fact I didn't see "why" using rotation (be able to better management of the strength for a printed part ? )

l1va, 2020-05-24:
Hello,
no, it is more like this one : [https://www.youtube.com/watch?v=I10bn53ECBo](https://www.youtube.com/watch?v=I10bn53ECBo)

fieldOfView, 2020-05-24:
For tilted conveyor bed, you can use Blackbelt Cura. Behind the scenes, this works by pretransforming the model before slicing it in the normal CuraEngine (I know, because I made the changes to support tilted gantries or beds in Blackbelt Cura).

For true 5 axis slicing, CuraEngine would need to be radically changed because as you say it relies on slicing along an straight axis.

l1va, 2020-05-24:
Thanks for the confirmation! 
If somebody thinks another way, let me know, please.

I will close the issue in some days.

Ghostkeeper, 2020-05-26:
Duplicate of https://github.com/Ultimaker/CuraEngine/issues/1256.

closed 2020-05-26 by Ghostkeeper