Ultimaker/Cura #7817: Unwanted solid infill causing bad overhangs

PuddingBaer91, 2020-05-24:
I already described the issue because it exists in all the popular slicers out there:

https://github.com/supermerill/Slic3r/issues/253

64Florian, 2020-05-24:
Hello,
You have a nice model,
Could you share you stl file in this ticket ?

For me it's "not" a bug because you ask for 10 layer as "bottom".
Take this product as sample:
[image]

[image]

Cura made what's you ask...

_But a new and grate feature could be adding the capability to change "bottom layers touching build plate"..._

-> In fact it's exist already under the name "initial bottom layers":
[image]

PuddingBaer91, 2020-05-24:
Sorry i didn't wanted to post it as a bug :D Sure i can share the model. It is from Extrudr:
https://www.thingiverse.com/thing:2838982

Ok for solid bottom surface layers i will play and tell you back my results. But for me bottom layers are layers starting from z=0 and ending at the reached number...

64Florian, 2020-05-24:
> bottom layers are layers starting from z=0

In fact not, in cura it's see as bottom as the product 

> i will play and tell you back my results
Thank you

PuddingBaer91, 2020-05-24:
Thank you :) I think 95% of the people out there thinking bottom layers are starting from z=0. I am printing since years and learned this truth just today :D:D but i also will try and report back. Thank you

64Florian, 2020-05-24:
For dev team, if you search for the same part, these are the rabbit teeth (start under the chin)

PuddingBaer91, 2020-05-24:
seems to work to set the initial bottom layers to 5 and the bottom layers to zero. Thank you so it seems to be that cura is the only one who is able to do that.

[image]

64Florian, 2020-05-24:
> the bottom layers to zero

You can, but may be not the best for you print (for me)... (_some gaps may appears_)
Just for your information (and in case), check to be sure "Outer before Inner walls" is disable too for this model

PuddingBaer91, 2020-05-24:
Thank you for the tipps. Here is the result with cura 0 bottom layers 5 initial bottom layers and 5 top layers: no gaps and perefect overhang. Only z seam are not so good (shortest and hiding) may you have a tip for get this better...

https://photos.app.goo.gl/7Zd9BVhLDmZvBoh78

Thank you for the hint with initial bottom layers option :) If you want i can report after some other prints if i have gaps or not. Atm i don't understand the sense of adding these bottom layers to the whole height...

64Florian, 2020-05-24:
> i don't understand the sense of adding these bottom layers to the whole height

Hmm, in fact hopefully you have another 'buggy" settings... 
But I was worried by "the idea" and my last test: 
> [image]

In your case Cura is "smart" (or have a nice bug, as you want): 
When you set to 0 you have this
[image]
Yellow = top /bottom .. but you ask for 0 -> The yellow part shouldn't be present
and layer after layer you have:
[image]
still Yellow ... 

In fact you shouldn't because you "ask" for NO bottom... 
Hopefully for you in your case...
In fact his bug can be forced / the workaround is to set the "main" setting "Top/Bottom Thickness" to 0 too. As you can see, it produce a "clean" layer without any "bad yellow" parts
[image]

-> this is Not what's "You" want... but in your case it make a nice work for you...
;-)

PuddingBaer91, 2020-05-24:
I understood the yellow was because you set top layers. So for me the yellow is top layers. If you disable them the yellow will go away

PuddingBaer91, 2020-05-24:
Ahhh now i got the thing with the gap. It's if we have to bridge. May there are also settings for bridging to set bottom bridge layers

64Florian, 2020-05-24:
Back for "Z" ..
Not sure ..But 
- This last piece is higher (at least 36 layer more as the "full gree") 
- This plastique isn't the same (this is back and the "full green" is bronze )

64Florian, 2020-05-24:
> I understood the yellow was because you set top layers. So for me the yellow is top layers. If you disable them the yellow will go away

Totally true ! I was focused with my filter !!
Thank you

PuddingBaer91, 2020-05-24:
> Back for "Z" ..
> Not sure ..But
> 
> * This last piece is higher (at least 36 layer more as the "full gree")
> * This plastique isn't the same (this one is back and the "full green" is bronze )

Don't understand what you want to say to me, sorry

64Florian, 2020-05-24:
Sorry,
Your last result : 
[image]

Has 36 layer more as the "green" one;
[image]

and you didn't use the same plastic (_plastique is the french work for plastic sorry_) 

In fact I didn't really understand your 
> Only z seam are not so good (shortest and hiding)

do you means seems or seam (sorry, for me seam is only on X-Y)

If it's about your seam, you could try this (_but not sure the result will please you_)
[image]

Hem..
Did you change your infill between the "Green" and you last test ? 
according:
> [image]
and your ticket supermerill/Slic3r#253

you switch from grid to Gyroid (and I know Gyroid is more "hard" for the surface)
-> you have to increase the number wall or tune "Infill overlap"...
[image]

**To be sure, the fast way to work is to disable infill an print again without touching any other settings.
**
-> it should be glorious. If not, tune the seam.
not sure it's the best for you, but it could made a nice job:
[image]

When you have something great, enable the infill. It could be better (or not) to made infill after your wall (in fact it depend of what's you want to print ... )
[image]

Sorry It's more difficult to clearly answer because it depend of your idea of a "good" print and depend of your object too :-(

PuddingBaer91, 2020-05-24:
Thank you i am stupid i forgot my own question about the z seam 😅 my filament ran out therefore i switched filament. I printed more layers because with cura i sadly can say: stop at layer/height X... So i just stopped manually too late. For me gyroid is the best infill. I noticed that i used a fixed z seam with the other slicer. So it's because the different z seam position...
Thank you for your large explanation and your help :)

64Florian, 2020-05-24:
> For me gyroid is the best infill.

For external result ? 
-> I'm not so sure ... 

For weight and resistance ? 
-> definitively Yes !

We discuss about "quallity" and evolution of the seam that's why I'm not so sure.

PuddingBaer91, 2020-05-24:
If you print outline before infill you will not see the gyroid wiggle on the outside walls

Ghostkeeper, 2020-05-26:
Try increasing the [Skin Removal Width](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/infill/skin_preshrink.md). This is what it's made for.

The discussion in this thread is a bit hard to follow I must admit. What is the change that you'd like us to make? Could you fill out the issue template?

PuddingBaer91, 2020-05-26:
@Ghostkeeper thank you. The suggestion to increase the Skin Removal Width is also a solution. So because my problem is solveable with the already existing features i will close this.

For all other people who will find it's way to this discussion: If you get bad overhangs try to look if your slicer has added skin to the overhang sections who are printed bad. If yes than try to play with the bottom/initial bottom layers or skin removal width.

Thank you guys :)

closed 2020-05-26 by PuddingBaer91