Ultimaker/Cura #7818: Overextrusion on thin line ceiling.

viesturz, 2020-05-24:
**Application version**
4.6.1, master 20200515

**Platform**
Win64

**Printer**
Custom

**Reproduction steps**
1. Slice a 2x100x2mm cube
2. Print it

**Screenshot(s)**
https://photos.app.goo.gl/BFj7MVuJkph3i14F7

**Actual results**
Severe overextrusion at the middle of the cube.

**Expected results**
Nice print.

**Project file**
[CFFFP_Test.zip](https://github.com/Ultimaker/Cura/files/4673936/CFFFP_Test.zip)

**Log file**
N/a

**Additional information**
This was not there a few versions back when the narrow ceiling was made using zigzag.

There is now an extra perimeter line from "extra skin wall count" that defaults to 1. Setting it to 0 restores previous behavior.

On very superfical level seems that "Compensate wall overlaps" should do the adjustment to fix this, but it does not do a thing.
The issue is very obvious in the GCode output - it extrudes 6 lines instead of 5 (2/0.4=5). With the middle two ones being on top of each other.

I tried disabling "fill gaps between walls", with no changes.

Made a 1.6-2.4mm wide test bits in 0.2 increments:
[image]

Printed result:
[image]

The 2mm one is way more overextruded than others with 1.4 and 2.2 also being slightly overextruded.

64Florian, 2020-05-24:
Hello,
As I can see in your project, you tune a lot of custom settings...
[image]
[image]
[image]

Could you save those modifications in a new profile* and select your draft to have a "clean" workspace ?

*To save your current profile, click on the around
[image]
[image]

> it extrudes 6 lines instead of 5 (2/0.4=5).
I'm a bit lost... just to be sure, each layer is (in draft as selected in your project) "only" 0.2 mm thick
For the width of this test (2mm width), it's well 5 lines with your 0.4mm nozzle: 
[image]

In your 
> Made a 1.6-2.4mm wide test bits in 0.2 increments:
> [image]

Could you confirm it's 1.2 mm to 2.6 mm ? 

In my point of view, if you save you settings and select draft again, you should have better results

-> As test just print the 2mm test 

Thank you

viesturz, 2020-05-24:
Hi,

Will test with the default settings and let you know.

Yes you are right about the 1.2-2.6mm, I cannot count :p

About 5 vs 6 lines: I was referring to the gcode being generated for a
single layer.
The visualization shows 5 lines per layer as expected, but the underlying
GCode for top layer contains two lines going down the middle at almost the
same coordinates, both running at full extrusion speed. Did not check the
lower layers though.

Thanks,
Viesturs

viesturz, 2020-05-25:
All default settings, except for fan speed.

Project: 
[CFFFP_Epic Habbi-Turing.zip](https://github.com/Ultimaker/Cura/files/4676924/CFFFP_Epic.Habbi-Turing.zip)

Result:
[image]

IMO the problem is obvious at GCode level and printing this out does not give much value:
> ;LAYER:8
;TYPE:WALL-INNER
;MESH:Epic Habbi-Turing.stl
G1 F2363.1 X-9.4 Y-.4 E0.02661
G1 X9.4 Y-.4 E0.62529    <- outer wall 1
G1 X9.4 Y0.4 E0.02661
G1 X-9.4 Y0.4 E0.62529   <- outer wall 2
G0 F7200 X-9.8 Y0.8
;TYPE:WALL-OUTER
G1 F1181.6 X-9.8 Y-.8 E0.05322
G1 X9.8 Y-.8 E0.6519    <- inner wall 1
G1 X9.8 Y0.8 E0.05322
G1 X-9.8 Y0.8 E0.6519   <- inner wall 2
G0 F7200 X-9.8 Y0.6
G0 X-9.01 Y0.01
;TYPE:SKIN
G1 F1181.6 X-9.01 Y-0.01 E0.00067
G1 X9.01 Y-0.01 E0.59935    <- centerline
G1 X9.01 Y0.01 E0.00067
G1 X-9.01 Y0.01 E0.59935    <- one more centerline, same amount extruded
G0 F7200 X-9 Y0.00

64Florian, 2020-05-25:
interesting ...
-> In fact Cura print the center as a wall too ... (unfortunately)
If you tune "wall line count" to 1 and infill 100% type line, you should have a good product .. 
(of course it's time consuming to use this setup )

Due to wall overleaping, when you ask for 0.4  you have 5% of overlapping (for my default settings)
so, for 2mm it's not 0.4x5 ... 
but the amount of materials look too much event it's less as regular external et internal wall  

@Ghostkeeper Have you see this issue in another recent ticket ?

Ghostkeeper, 2020-05-26:
I haven't, no. But I understand what the issue is from this:

> ;TYPE:SKIN
G1 F1181.6 X-9.01 Y-0.01 E0.00067
G1 X9.01 Y-0.01 E0.59935 <- centerline
G1 X9.01 Y0.01 E0.00067
G1 X-9.01 Y0.01 E0.59935 <- one more centerline, same amount extruded

Wall overlap compensation only works on walls, not on skin. Not even the extra concentric outline around skin.

In the original image, this only applies to the second and maybe the third rectangle from the left: 

> [image]

The leftmost rectangle doesn't have walls overlapping since the first inner wall didn't fit any more. All rectangles further right don't have walls overlapping because the walls are spaced far enough apart.

I think a solution would be to set the number of extra skin lines to 0 and add another wall instead.

I'm not sure if we want the Wall Overlap Compensation to apply to the extra skin lines and extra infill lines as well. We're trying to [replace the entire feature](https://github.com/Ultimaker/CuraEngine/pull/1210) actually.

Ghostkeeper, 2021-12-14:
This is fixed in https://github.com/Ultimaker/CuraEngine/pull/1210.

closed 2021-12-14 by Ghostkeeper