Ultimaker/Cura #7819: No Experimental tree wall thickness 4.6.1

n5vei, 2020-05-24:
I am using 4.6.1 Cura. I used trees for support material before upgrading to 4.6.1.  Now when I use tree support, the tree wall thickness is extremely thick!  I have all boxes showing when I try to find wall thickness, but it is not there.  I cannot use tree support because of the thickness.

64Florian, 2020-05-24:
Hello,
Could you share your workspace please ? 
Open your project with an object inside, then select "Flie" -> "save" -> "save" -> provide a filename -> select "open folder" 
Zip this your new "3mf" file and add it in this ticket

Thank you

n5vei, 2020-05-24:
[CCR10S_Pilgrim_City_House_C_Back1.zip](https://github.com/Ultimaker/Cura/files/4674597/CCR10S_Pilgrim_City_House_C_Back1.zip)

closed 2020-05-24 by n5vei

Ghostkeeper, 2020-05-26:
The wall thickness of that support seems pretty normal to me:

[image]

If it prints wrong, try adjusting the Support Line Thickness setting.