Ultimaker/Cura #7820: Pound symbol in filename causes OpenRecentFile menu errors

KenLucke, 2020-05-24:
Cura 4.6.1
MacBookPro
Mac OS X 10.15.4 (latest)

If a filename has a pound symbol ("#") in the filename, it gets translated to "%23" in the Open Recent File menu.

For instance, a file saved with the name

`"Puzzle Chessboard Pieces #2 (x16).3mf"`
gets translated to:
`"Puzzle Chessboard Pieces %232 (x16).3mf"`

As for some reason, github will not alllow me to upload a file, here is a link to an image on Dropbox:
https://www.dropbox.com/s/7q2u409y490ih9w/OpenRecent%20Bug.jpg

This is only _cosmetic in the menu_.  The file opens properly, and retains the proper filename while loaded, and for saving.  The normal file open dialog (being OS created) is not affected.

64Florian, 2020-05-24:
Hello,
I reproduce this issue on my "old" 10.12.6 MacOS with 4.6.1, 4.2.1 and 2.7.0 too 
@KenLucke Do you remember if you saw a version of Cura properly working with this feature ?

KenLucke, 2020-05-24:
> Hello,
> I reproduce this issue on my "old" 10.12.6 MacOS with 4.6.1, 4.2.1 and 2.7.0 too
> @KenLucke Do you remember if you saw a version of Cura properly working with this feature ?

No, I cannot recall if it happened previously.  I rarely use the pound sign in file names, so I probably never experienced it before, especially as it only hits in the Recent menu.

Ghostkeeper, 2020-05-26:
Yeah, reproduces for me as well. Good find!

nallath, 2020-05-27:
Fixed this for the next release (4.7)

commit b58730a7 (2020-05-27): Fix decoding in recently opened menu

Fixes #7820
closed 2020-05-27 by nallath