Ultimaker/Cura #7821: Cura 4.6.1

Kaicura, 2020-05-25:
Cura 4.6.1
Egal was ich bei Fluss einstelle, ea ändert sich nichts an der Wandstärke.
Testwürfel mit 2 Wänden a 0,42 mm bei 100 % Fluss = 0,76 mm breite
Bei 150% Fluss auch 0,76mm breite
Habe mal das Profil mit 100% Fluss hochgeladen

[Test_Profil_CUBE.zip](https://github.com/Ultimaker/Cura/files/4674867/Test_Profil_CUBE.zip)

mahtDFR, 2020-05-25:
I'm seeing this when I open your project file.
[image]

How did you add this machine definition?

Kaicura, 2020-05-25:
It is from Site 3D nexus
, Artillery Sidewinder x1 cura Profil s and Printer and Nozzel definations 
--
Gesendet über myMail für Android Montag,

Kaicura, 2020-05-25:
Here the Files for CURA 4.6.1 
[3DN Artillery SWX1 and Genius Cura 4.6.1.zip](https://github.com/Ultimaker/Cura/files/4678687/3DN.Artillery.SWX1.and.Genius.Cura.4.6.1.zip)

Import the  Cura profils at Profilsmanager, Copy the  Folders to Cura / resources

Ghostkeeper, 2020-05-26:
The reason is that your object overrides the Flow setting in its per-object settings.

[image]

The per-object settings override the global (or per-extruder) settings.

closed 2020-05-26 by Ghostkeeper

Kaicura, 2020-05-26:
Thanks a lot for your great and fast support