Ultimaker/Cura #7823: Add definitions for Makerbot Replicator 2 with metadata to allow easy X3GWriter export

JesseCake, 2020-05-25:
The Makerbot Replicator 2 was a very popular printer (at least it was here in Australia) with many many institutions and schools buying it for their maker spaces.

Now that Makerbot is making it increasingly difficult for users to use these still very well maintained machines (latest builds don't appear to print very well, Makerware is no longer supported for Linux etc) it'd be great if Cura allowed these users to quickly and easily jump over and continue to use them rather than retire the hardware.

There is a definition for a Replicator 1 printer built in, but it lacks the metadata to pass to X3GWriter to convert to the native X3G required to drive the printer. Of course users can create a new profile and edit the metadata to make this happen, but this may be beyond the scope of the average user. 

Without this metadata, the prints come out the wrong size due to step numbers etc difference between Replicator 1 and 2.

I blogged about methods to make manual hacks to X3GWriter to make it always convert to Replicator 2 some time ago and get comments every week with people asking questions about it. It isn't the ideal situation doing it this way though.

It would be as simple as creating another profile built into Cura to choose from the menu for Replicator 2 with metadata including:
"machine_x3g_variant": "r1"

Along with of course the correct depth width etc sizing required. I can have a go at creating this profile, but I've never contributed to someone else's project on github before, only my own. I don't know the protocol to do so.

JesseCake, 2020-05-25:
Ok I have a basic definition here - hopefully it's enough to get a definition started:

`{
    "name": "MakerBotReplicator2",
    "version": 1,
    "inherits": "fdmprinter",
    "metadata": {
        "visible": true,
        "author": "Ultimaker",
        "manufacturer": "MakerBot",
        "machine_x3g_variant": "r2",
        "file_formats": "application/x3g",
        "platform_offset": [ 0, 0, 0],
        "machine_extruder_trains":
        {
            "0": "makerbotreplicator_extruder_0"
        }
    },

    "overrides": {
        "prime_tower_size": { "default_value": 10.0 },
        "infill_sparse_density": { "default_value": 10 },
        "layer_height": { "default_value": 0.15 },
        "machine_extruder_count": { "default_value": 1 },
        "machine_heated_bed": { "default_value": false },
        "machine_center_is_zero": { "default_value": true },
        "machine_height": { "default_value": 150 },
        "machine_gcode_flavor": { "default_value": "MakerBot" },
        "machine_depth": { "default_value": 145 },
        "machine_width": { "default_value": 225 },
        "machine_name": { "default_value": "MakerBot Replicator 2" }
    }
}
`

I also have screenshots attached here for my interface settings that should probably go into this definition, but unsure how to incorporate into the JSON structure you have going on in Cura.

Lastly, I have custom gcode that is required to make the printer function as without it it fails to print. Can I add this to the printer definition so that it's nice and straight forward?

**Under printer start Gcode:**

`; -- start of START GCODE –
M73 P0 (enable build progress)
;M103 (disable RPM)
;G21 (set units to mm)
M92 X88.8 Y88.8 Z400 E101 ; sets steps per mm for replicator
G90 (set positioning to absolute)
(**** begin homing ****)
G162 X Y F4000 (home XY axes maximum)
G161 Z F3500 (home Z axis minimum)
G92 Z-5 (set Z to -5)
G1 Z0.0 (move Z to "0")
G161 Z F100 (home Z axis minimum)
M132 X Y Z A B (Recall stored home offsets for XYZAB axis)
(**** end homing ****)
G92 X147 Y66 Z5
G1 X105 Y-60 Z10 F4000.0 (move to waiting position)
G130 X0 Y0 A0 B0 (Set Stepper motor Vref to lower value while heating)
G130 X127 Y127 A127 B127 (Set Stepper motor Vref to defaults)
G0 X105 Y-60 (Position Nozzle)
G0 Z0.6     (Position Height)
; -- end of START GCODE –`

**And for the printer end Gcode:**

`; -- start of END GCODE –
G92 Z0
G1 Z10 F400
M18
M104 S0 T0
M73 P100 (end  build progress)
G162 X Y F3000
M18
; -- end of END GCODE –`

Happy to do the legwork as required, I'm just not all over the file structure for Cura to be happy.

mahtDFR, 2020-05-25:
We wont be making this ourselves, but we're happy to review pull requests if someone submits one. Please make sure it's fully tested!

JesseCake, 2020-05-25:
No problem, I'll test it out and see if I can get a coherent definition for you

Ghostkeeper, 2020-05-26:
You'd also need to create extruder definition files in the `extruders` resource folder.

wcarr92, 2020-05-28:
Hello.

I am trying to bring a Replicator2 into the Cura world as well.  I have managed to get the start gcode "machine_start_gcode" into the .def.json file as well as the end gcode "machine_end_gcode".  However, the value "Makerbot" for "machine_gcode_flavor" is being ignored in the .def.json file.  I'm getting "Marlin" flavor.  I can go into the machine settings from the UI and set the flavor to "Makerbot".  That change will appear in the "MakerBotRplicator_settings.inst.cfg" file and will output the proper gcode.  

The problem manifests itself in the release 4.6.1 from Cura.  If I select the MakerBot MakerBotReplicator, where within the released printer definitions, the following can be found:

    "overrides": {
        "prime_tower_size": { "default_value": 10.0 },
        "infill_sparse_density": { "default_value": 10 },
        "layer_height": { "default_value": 0.15 },
        "machine_extruder_count": { "default_value": 1 },
        "machine_heated_bed": { "default_value": true },
        "machine_center_is_zero": { "default_value": false },
        "machine_height": { "default_value": 150 },
        "machine_gcode_flavor": { "default_value": "MakerBot" },
        "machine_depth": { "default_value": 145 },
        "machine_width": { "default_value": 225 },
        "machine_name": { "default_value": "MakerBot Replicator" }
    }

However, the generated gcode still indicates flavor "Marlin".  I don't think asking the user to go into the printer settings to set flavor "Makerbot" should be necessary.

wcarr92, 2020-05-29:
Answered my own question.  The "machine_gcode_flavor" needs to have a value "Makerbot"  NOT 
"MakerBot".  Notice the lowercase "b" in Makerbot.

chepo92, 2020-06-17:
@JesseCake , not directly related to the "issue" but i would recommend flashing Marlin firmware to your printer

JesseCake, 2020-07-22:
This sounds incredibly interesting, and I have 3 so I can test this with
one. Do you have any direct experience with doing this process (flashing to
Marlin)? So far all I can find is people discussing it and explaining the
hardware changes they think would need to happen to make it work..

Jesse

chepo92, 2020-07-22:
@JesseCake 
Yes I have experience on it, as it's kind of a hobby and a part-time job in my entrepreneurship, I have a FlashForge 2x (the wooden one) running Marlin.
You won't find anything in forums as not many people have noticed that it can be done now (with marlin 2), just needs to spread the word. You may read directly on github issues for some info (probably you will find issues I have submitted)
I know by experience it can be done to any Mightyboard-like board (rev A - Rev H), including brands as Makerbot, flashforge, qidi, Mbot, etc, i have contributed those config to marlin repository. But not with newer Makerbot versions (those are closed source, and would need some time/work to make marlin run on it, but it's just a matter of time)

Mcudnic, 2023-02-13:
Basically the Flashforge creator pro is the same machine. At my work i still had a rep 2x working horribly. Got tired of trying to make it work as i was not knowledgeable enough to fiddle  with the defintionion files. So i tore out the mightyboard, inside and replace it with Ramps 1.6 for the simply reason that i need an extra fan to print pla. Then i simply added a custom printer and i works perfectly i cura. I'm also familliar with Marlin 2.09, so this was a no brainer for me personally. I did this conversion a other printer as well. Then there was a sale for the Flashforge so i bought one for a mere 400 usd. Only to disover it was the same machine really. Flashprint is better than makerbots slicer, but the support feature is total bullshit. But Toylerr make the defintions files on it. So if this is still on your radar ! go check it out. No i can print pla and use either pva for support. And also nylon230. Awesome prints for nylon and pva ! I did some small details for an rc car for example. I work at a school as a helper between teacher and students, plus being the goto guy the get stuff printed of course

closed 2024-02-27 by github-actions[bot]