Ultimaker/Cura #7825: Cura 4.6.1 crashes when saving gcode

aguiars, 2020-05-25:
Cura crashes every time when saving .stl file to gcode. I already unchecked send anonymous print information under preferences as this has worked in previous versions. Does this have anything to do with not installing extra software during initial Cura install? I have attached a copy of the logs. Please help.

**Application version**
Cura 4.6.1

**Platform**
Windows 10

**Printer**
Creality 3 Pro

[cura.log](https://github.com/Ultimaker/Cura/files/4677530/cura.log)

mahtDFR, 2020-05-25:
Thanks for the log. According to this it is slicing, at least sometimes:

``2020-05-25 09:41:28,756 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [678]: Slicing took 3.85392165184021 seconds``

Are your graphics drivers up to date? At the end of your log I can see this:

``2020-05-25 09:56:58,754 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [116]: GLSL Version:    4.10.0
2020-05-25 09:56:58,764 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\default.shader]...
2020-05-25 09:56:58,803 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\selection.shader]...
2020-05-25 09:56:58,843 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\select_face.shader]...
2020-05-25 09:56:58,874 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\default.shader]...
2020-05-25 09:56:58,893 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\composite.shader]...
2020-05-25 09:56:58,912 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\overhang.shader]...
2020-05-25 09:56:58,932 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\striped.shader]...
2020-05-25 09:56:58,950 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\transparent_object.shader]...
2020-05-25 09:56:58,977 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\striped.shader]...
2020-05-25 09:56:58,994 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\xray.shader]...
2020-05-25 09:56:59,017 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\xray_composite.shader]...
2020-05-25 09:56:59,040 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\default.shader]...
2020-05-25 09:56:59,057 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\grid.shader]...
2020-05-25 09:56:59,108 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\platform.shader]...
2020-05-25 09:56:59,134 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\xray.shader]...
2020-05-25 09:56:59,770 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [187]: Error check finished, result = False, time = 13.7s``

aguiars, 2020-05-25:
I just updated drivers through a windows update check and the problem still occurs.

Ghostkeeper, 2020-05-27:
> Does this have anything to do with not installing extra software during initial Cura install?

Probably not. Unless you're printing via USB or updating printer firmware via USB, the Arduino drivers are not involved. And the C++ Runtime hasn't changed since Cura 2.1 as far as I know so if you've installed that for any version of Cura you will have it installed for all of them.

Your log file doesn't contain any information about a crash, but you can see that the program does indeed not close properly. To me this looks like a segfault. Do you see any kind of crash report, or pop-up, when it crashes? Does the program freeze, maybe?