Ultimaker/Cura #7828: Brim Depth

kfsone, 2020-05-25:
The addition of the brim-distance feature is outstanding but it also suggests a secondary feature that would be super helpful: horizontal support for stability when printing thin vertical features, as the blades in the attached model.

This could be achieved by having an option to continue the brim for additional layers.
[Wahl__Hair_Attach_19mm_Six.zip](https://github.com/Ultimaker/Cura/files/4678602/Wahl__Hair_Attach_19mm_Six.zip)

Beyond the initial layer, this helps stabilize the model in the X and Y directions. Additional settings to allow the brim to be further from model would assist in post-print separation

Example use with the attached model:
```
Build Plate Adhension Type: Brim,
Brim Width: 10mm,
Brim Layer Height: .3mm
Brim Distance: .35mm
Brim Height: 9mm,
+ Brim Layers: 30,
+ Brim Taper: Yes/No
+ Brim Top Width: 3mm
+ Brim Top Distance: .4mm
```

When Brim Taper is enabled, the width/distance morphs between values; otherwise the brim adopts the 'top' values after the first layer.

Lastly, perhaps:

```
+ Maximum brim support thickness:
```

This would allow you to specify that the brim only be retained within Nmm of parts this thick: if you were printing a triangle 200mm long, 25mm at the widest end, a max brim support thickness of 5 would continue brim at the points of the triangle and along the portions where the model is <= 25mm across.

rburema, 2020-05-27:
Have you tried out how this might effect the print in practice? (By editing gcode for example?)

Ghostkeeper, 2020-05-28:
Duplicate of https://github.com/Ultimaker/Cura/issues/6939.

closed 2020-05-28 by Ghostkeeper

mporter88, 2021-03-22:
I believe that an option that created a brim that was an extension of the bottom layers. and allowed tapering down to one layer, and REQUIRED removal by mechanical methods would be a valuable addition to cura.
Such tools are not that exotic.  Certainly not in a production operation.

Rafts do not do the same thing for me as creating a larger footprint does.  Difficult materials
will sometimes pull from a raft while they do not when I design a brim that tapers down in layer count.

Creating a more rigid footprint is extremely effective...and cutaway is simple with a laminate trimming tool.

I submitted this at pull #9426 with a detailed argument for the options proposed... and it was labelled a duplicate.