Ultimaker/Cura #7829: FlashForge Finder support

ronoaldo, 2020-05-25:
**Is your feature request related to a problem? Please describe.**
I would like to add support for FlashForge Finder in Cura. I did a little scripting already and have some pieces that I would love to put together. I own a FFF and it is a good starter printer. I use Cura with it already but did several hacking to get some things working, so having a few of them ready from Cura itself would be awesome.

**Describe the solution you'd like**
I would love to see Cura support for Finder as a Machine Definition built-in, so users can get it ready to use more easily. I can create the definition with the data I used already in the UI so it would be cool, just need proper directions on how to proceed.

Also, would like to support some features like sending the job over wifi. Today I operate by uploading the g-code from Cura to a cloud provider as I don't have a USB stick (odd, i know!). Then I use the built-in firmware feature to download from cloud and print. I would love to implement the functionality to send the print job over wifi and keep monitoring it over as the printer allows it using TCP. I have some pieces like a custom GX code writer that the printer uses and allows a print preview image to be displayed in LCD (testing yet).

Finally, I have a small script to work around the "Pause at height" solutions from default Cura that oddly don't work with Finder (printer goes nuts if I use them). I implemented a small script to allow to use the M25 command that the printer understands. Would like to ship it (or a version of it using already existing plugins) to avoid a set of trial and error for other users.

**Describe alternatives you've considered**
I am using these for a year and think this would be a good addition for an upcoming Cura release, as I often find users on Forums that don't know how to get the job done with Cura and may be losing a lot of good features from it.

**Affected users and/or printers**
Only owners of FlashForge Finder that don't like/can't use the provided Slicer software.

**Additional context**
I need some guidance in order to get things done but I'm all in to send pull requests and get feedback/review. I also can code in Python and have the printer so I can test different things until we get it ready to release.

ronoaldo, 2020-05-26:
I just now realised I can't even view the machine settings anymore ... I have all AppImage I used so far, so I could roll back and look that last version working was 4.3, since 4.4.1 if you add a "custom fdm printer" it just don't let you edit any settings. This makes adding the printer definition even more important.

mahtDFR, 2020-05-26:
This is a good starting point:
https://github.com/Ultimaker/Cura/wiki/Adding-new-machine-profiles-to-Cura

`Also, would like to support some features like sending the job over wifi. Today I operate by uploading the g-code from Cura to a cloud provider as I don't have a USB stick (odd, i know!). Then I use the built-in firmware feature to download from cloud and print. I would love to implement the functionality to send the print job over wifi and keep monitoring it over as the printer allows it using TCP. I have some pieces like a custom GX code writer that the printer uses and allows a print preview image to be displayed in LCD (testing yet).`

Some other printer manufacturers are doing something similar, so perhaps you could look at their codebase for inspiration, e.g. https://github.com/HydraResearchLLC/Nautilus-Cura-Plugin

eskeyaar, 2020-05-26:
> I just now realised I can't even view the machine settings anymore ... I have all AppImage I used so far, so I could roll back and look that last version working was 4.3, since 4.4.1 if you add a [...]

> **Is your feature request related to a problem? Please describe.**
> I would like to add support for FlashForge Finder in Cura. I did a little scripting already and have some pieces that I would love [...]

I think I can help in adding finder defjson if you can give me proper start gcode, end gcode, and other things. so that we can move forward it.

ronoaldo, 2020-05-26:
> This is a good starting point:
> https://github.com/Ultimaker/Cura/wiki/Adding-new-machine-profiles-to-Cura

I saw the docs at the wiki, thanks! I'll follow them. The docs don't state, say, how I can test my definitions in my local install... I use the .AppImage on Linux. Is there a way for me to test them before sending the Pull Request?

> `Also, would like to support some features like sending the job over wifi. Today I operate by uploading the g-code from Cura to a cloud provider as I don't have a USB stick (odd, i know!). Then I [...]

This is amazing! I'll check it out and follow their codebase as a sample. Thank you!

ronoaldo, 2020-05-26:
> I think I can help in adding finder defjson if you can give me proper start gcode, end gcode, and other things. so that we can move forward it.

Cool! I'll downgrade to 4.3 and open the printer settings dialog to share here the details, but can you check if this old post from myself on Reddit can be used as a starting point?

https://www.reddit.com/r/FlashForge/comments/cych1o/flashforge_finder_profile_for_cura_4/eyt7rbx/

I can do test prints with a defjson once I understand how to test them in my local install.

Ghostkeeper, 2020-05-27:
> Is there a way for me to test them before sending the Pull Request?

Yes. In Cura, click on Help -> Show configuration folder. You'll be presented with two folders then and one of them contains a number of subfolders like "definitions", " extruders", etc. That is where you can save any custom profiles. At the very least you'd need to add one definition and one extruder. You can look for examples in the resources folder in this repository, and read the documentation on that wiki.

eskeyaar, 2020-05-27:
> Cool! I'll downgrade to 4.3 and open the printer settings dialog to share here the details, but can you check if this old post from myself on Reddit can be used as a starting point? [...]

Hi,
i just made definitions and extruders def.json for FF Finder. 
https://github.com/eskeyaar/Flashforge-Finder-
Download those def.json and paste them in respective folders and play with some settings and see how it work.

Used another source of start gcode : https://labs.tomasino.org/flashforge-finder-with-cura/

Let me know if this works for you. 
Thanks

ronoaldo, 2020-05-27:
Hi! Thanks for the updates!

@eskeyaar I'll test later tonight. About the edit, i probably need to clean up all my configuration then, probably some odd config from older versions are preventing me to edit the machine settings.

eskeyaar, 2020-05-27:
> 
> 
> Hi! Thanks for the updates!
> 
> @eskeyaar I'll test later tonight. About the edit, i probably need to clean up all my configuration then, probably some odd config from older versions are preventing me to edit the machine settings.

Your saying you cant edit any of machine settings in cura? maybe try a clean install again.

ronoaldo, 2020-05-28:
@eskeyaar thanks for the info! Yeah, I was not cleaning up properly. Some odd move between different versions of cura probably broke my setup. After removing both configuration folders it worked just great.

Used this clean setup to test the def.json files: they also worked perfectly. I sliced a benchy and now it is printing! I could not find the right way to import the profile, used the default ones but I'll take a look at the settings you provide as well.

I would suggest removing the "xgcode" part from start gcode. The file used by Finder is basically regular g-code, but Flashprint provides a special two line header in binary format. That part is where I plan to allow the printer to recognize the print time, temperature and show a nice thumbnail. I'm working in the .gx writer here https://github.com/ronoaldo/cura-flashforge-finder, based on work from @Ghostkeeper.

I'll update once benchy is done and I do some edge testing (full area test, max height test).

I would like to reconsider my request and break into two wafes, one for offline printing improvement and another for the online part, as I need to go further in that direction.

Would it be possible to inclue:
1. The def.json printer and extruder information as they seems to work
2. The script to pause at height (FlashForge Finder)
3. Add this STL and maybe this image as the printer design for showing in Cura (https://www.thingiverse.com/thing:2999750)

The other feature for print job monitoring would require me to go extra miles so I'll work on that later on.

ronoaldo, 2020-05-28:
First 3d print was a success! I did some testing with the Pause at layer script and it also generated a valid g-code. I tested the GXWriter work in progress and it could be loaded in the printer properly, it recognized the print time from Cura and displayed a placeholder image. Now I need to work on the image processing to embed the print preview and it will be amazing.

Few things I noted in my setup. The "Print head section" I filled blindly, not sure how it affects print time or anything like that. I read it more carefully and it expects me to enter values for the print head dimensions? How that impacts the generated gcode?  How I proper measure them in my printer so we can place appropriate values in the def.json printer file?

Also, I noted that if I load an object that fill the full build plate horizontally (140x140x1), I can't slice it. It requires me to lower the print to 99%. Is there a way to "fix" that? I tested adding 2 more millimetres and it allowed me to slice, printing a test now to make sure it does not barfs in the printer itself. Note: I disabled the bed adhesion so it should work properly. Not sure if we can tell Cura that there is a small tolerance around the print bed.

Final question, is it possible to use PIL in my extension/plugin?

eskeyaar, 2020-05-28:
> @eskeyaar thanks for the info! Yeah, I was not cleaning up properly. Some odd move between different versions of cura probably broke my setup. After removing both configuration folders it worked [...]

Great! Regarding my test profile, i used S3D PLA settings and Flashprint to complie. So it should be much better than default profile i guess. 

>  1. The def.json printer and extruder information as they seems to work

What information do you want me to include?

>     3. Add this STL and maybe this image as the printer design for showing in Cura (https://www.thingiverse.com/thing:2999750)
Will add them, and also removed xgcode from start gcode script.

> Also, I noted that if I load an object that fill the full build plate horizontally (140x140x1), I can't slice it. It requires me to lower the print to 99%. 

i think it has to do with Printhed settings (X min Y Min etc..) under machine settings, let me see what i can make changes for that. 

>  How I proper measure them in my printer so we can place appropriate values in the def.json printer file?
Not sure on how to do that? if someone can help that, i will modify it. only thing in creating def.json values for those i struggle lot, so confusing on those coordinate values.

Ghostkeeper, 2020-05-28:
> Final question, is it possible to use PIL in my extension/plugin?

Pillow is not shipped with Cura (and PIL neither since that is only Python2). You can include it in the plug-in folder though, if you like. Since Pillow is a binary plug-in though, you'd need to include 2 versions of the library: one for Windows, one for MacOS and one for Linux. Then you need to load the correct version in the Python import, depending on the user's OS.

For an example, I've done that as well for the SVG Toolpath Reader:
* The plug-in distribution contains a version of Freetype for [MacOS and Windows](https://github.com/Ghostkeeper/SVGToolpathReader/tree/master/freetype) (but not Linux since most Linux distributions come with this library pre-shipped).
* libFreetype is imported [depending on the operating system](https://github.com/Ghostkeeper/SVGToolpathReader/blob/9354c8223045b4996066c72b5b21f0ad7c410b97/freetype/raw.py#L24-L30) (not my code in this case).
* Since plug-ins aren't imported relative to the plug-in folder but relative to Cura's root, you also need to [import manually using importlib](https://github.com/Ghostkeeper/SVGToolpathReader/blob/a2bbe90da64e6cd9d54fec553f61ba941001e85d/Parser.py#L26-L34).

Ghostkeeper, 2020-05-28:
You can also consider using PyQt (look for the Qt documentation) which also has image reading functionality. We're using Qt5.10 at the moment.

eskeyaar, 2020-05-28:
> First 3d print was a success! I did some testing with the Pause at layer script and it also generated a valid g-code. I tested the GXWriter work in progress and it could be loaded in the printer [...]

Made my changes on def,json over build size and added mesh of Finder base.
The build size now can able to load 140mm x 140mm, but i did increased overall print bed size over 141.3mm. Not sure why 140mm not able to slice when bed size is 140mm.

ronoaldo, 2020-05-29:
Ok, I guess this is coming along nicely!

First, the new version of the printer def.json file is pretty neat. The final build-plate mesh touch was fantastic, thank you! And the extended edges allow me to use full build size. :+1: 

[image]

Second, the profile is perfect!!! I usually lower the default Flashprint temps for PLA from 220 down to 200, so strings and ooze disappear, but the overall print quality is noticeably better. It could be my filament too, I guess Flashforge filament needs the 220 temp, so it would probably make sense to keep it that way.

[image]
*The blue one uses the provided profile and the white default 0.2mm profile*

Third, following the suggestions from @Ghostkeeper with the QImage from PyQt5 I could complete the .gx header preview image logic and i now got a supercharged Cura for my Finder! I had to share this pic:

[image]

The reason I put some effort here is that the finder UI makes it pretty easy for my family to use the print jobs I already have sliced, so it is very good to have this information.

With this final testing all that is remaining is the network part but that would require a more advanced work and consume more time. With what I got so far I'm very happy already! :smile: 

To close this one, I would like to hear what is the best approach for each deliverable. The def.json files, if possible to be included into main distribution, would make a lot of sense.

The GXWriter I adapted is fully optional but completes the user experience. Ideally, they both would be used together so when you are using Finder the output defaults to .gx instead of .gcode and the GXWriter is used. Not sure if this can be achieved, but it is not a hassle to select .gx when saving if you want to. I could send my code to @eskeyaar if he wants to bundle a plugin, or the other way around: I can warp up all files as a plugin. I am member of a Facebook 3d printing group and know some people that could test the plugin on Windows to make sure a bundled version is good to go.

Really appreciate the effort from all of you into this request and await further instructions on how we can release this!

Ghostkeeper, 2020-05-29:
> await further instructions on how we can release this!

I think the best approach is to take a similar route as to how the Sailfish-based printers are working with the X3GWriter plug-in. The Makerbot Replicator, for instance, specifies that it prefers to use the X3G format as output:

https://github.com/Ultimaker/Cura/blob/5e73412a67a94a5acd12edb60bb28331c2be40e4/resources/definitions/makerbotreplicator.def.json#L10

You need to specify your format's MIME type there, `application/xgcode`.

Then you can submit your plug-in to the [Contributor Portal](https://account.ultimaker.com/) in order to get it published to the Marketplace, and your printer definition as a pull request to this Cura repository itself. If the plug-in is installed, your printer will automatically use .gx, and otherwise it will default to .gcode. The "pause at layer" script would need to be merged in with the current Pause At Height post-processing script, since we recently merged all of those flavours together into one script for clarity.

eskeyaar, 2020-05-29:
> 
> 
> Ok, I guess this is coming along nicely!
> 
> First, the new version of the printer def.json file is pretty neat. The final build-plate mesh touch was fantastic, thank you! And the extended edges allow me to use full build size. 👍
> 
> [image]
> 
> Second, the profile is perfect!!! I usually lower the default Flashprint temps for PLA from 220 down to 200, so strings and ooze disappear, but the overall print quality is noticeably better. It could be my filament too, I guess Flashforge filament needs the 220 temp, so it would probably make sense to keep it that way.
> 
> [image]
> _The blue one uses the provided profile and the white default 0.2mm profile_
> 
> Third, following the suggestions from @Ghostkeeper with the QImage from PyQt5 I could complete the .gx header preview image logic and i now got a supercharged Cura for my Finder! I had to share this pic:
> 
> [image]
> 
> The reason I put some effort here is that the finder UI makes it pretty easy for my family to use the print jobs I already have sliced, so it is very good to have this information.
> 
> With this final testing all that is remaining is the network part but that would require a more advanced work and consume more time. With what I got so far I'm very happy already! 😄
> 
> To close this one, I would like to hear what is the best approach for each deliverable. The def.json files, if possible to be included into main distribution, would make a lot of sense.
> 
> The GXWriter I adapted is fully optional but completes the user experience. Ideally, they both would be used together so when you are using Finder the output defaults to .gx instead of .gcode and the GXWriter is used. Not sure if this can be achieved, but it is not a hassle to select .gx when saving if you want to. I could send my code to @eskeyaar if he wants to bundle a plugin, or the other way around: I can warp up all files as a plugin. I am member of a Facebook 3d printing group and know some people that could test the plugin on Windows to make sure a bundled version is good to go.
> 
> Really appreciate the effort from all of you into this request and await further instructions on how we can release this!

Hi,
Great to hear it works! Thanks for the picture and feedback.
From the picture i see some stringing, so i have made little bit changes on retraction and updated the profile too. 
As @Ghostkeeper said, we can make this as similar to x3g, that would be a good one for everyone to use it. @ronoaldo Is your gxwriter for the plugin is completed? i can change the MIME type to gx.

ronoaldo, 2020-05-29:
@Ghostkeeper thanks, that would be very cool then! Just to make sure I got it right: we would bundle the a) eskeyaar .def.json files, b) my python implementation gxwriter as a plugin, say, FlashforgeFinderPlugin, and with the change to the mime type it will prefer the .gx output when saving to file after a slice done, correct? Just to make sure both the printer def.json and the writer are available at the end of the process.

@eskeyaar I'll check the updated profile, I really apreciate that. Best prints ever I'm pulling with it, want to test more. Is it possible to bundle this profile with the plugin too? To me this would be HUGE for users, as it does deliver very good quality print.

About the gxwriter, it is complete. It has several fallbacks, such as, bundle a "default" image if rendering the screenshot breaks, and fallback to standard gcode output if any of the gx header manipulation fails. Should be good to get the printer using it as default. The picture of the bench at the printer is it printing the full suite: used your def.json + gxwriter plugin + print profile to generate the .gx file and loaded it with USB (I found one at work that I managed to format and use yay!). Pretty neat workflow and worked well.

Also, I added the "Pause at layer (Flashforge Fidner)" post-processing script to my repo... is this the correct way to deliver that one? Sadly, I tested all the available post-processing and this solution worked best for the Finder, as it adds the  M25 instruction just like Flashprint does and the printer firmware handles that nicely.

eskeyaar, 2020-05-29:
> @Ghostkeeper thanks, that would be very cool then! Just to make sure I got it right: we would bundle the a) eskeyaar .def.json files, b) my python implementation gxwriter as a plugin, say, [...]

@ronoaldo Yes, we can bundle this profile along with plugin and machine definitions. Just keep testing and let me know if something is need to change in profile, or i can change based on the picture of issue. 
I also tried your gxwriter by pasting and tried to slice, couldn't see .gx anywhere? maybe im doing wrong. 
even tried to change MIME type to gx instead of xgcode. still not able to get it. maybe i should check other file too. 
Thanks,

ronoaldo, 2020-05-29:
@eskeyaar thanks for the heads up. I renamed the repo to https://github.com/ronoaldo/FlashforgeFinderIntegration following the naming guidelines. It should be shown as one of the file types when saving after slice is complete. If the plugin can't be loaded (say, because I had a syntax error or used modules unavailable somehow) it would not show, and a stacktrace will be available at stderr.log under the config dir.

I'll try again with another machine and another cura version to make sure I hasn't break anything last night :see_no_evil:

ronoaldo, 2020-05-29:
OK, updated the repo again, this time with the lower SDK version at plugin.json. Kindly let me know if that works.

eskeyaar, 2020-05-29:
OK, cura 4.0 works, but doesnt work at new versions:
try adding this "supported_sdk_versions": ["5.0.0", "6.0.0", "7.0.0", "7.1.0"] this worked for my old and newer version of cura.

eskeyaar, 2020-05-29:
@Ghostkeeper how do I publish my .curaprofile along with machine definition? 
Thanks

ronoaldo, 2020-05-29:
@eskeyaar thanks for the PR, merged in! Testing some prints now ... 

[image]
*Added Gnome preview support for the GX files so it is pretty nice to view these*

ronoaldo, 2020-05-30:
Small update. I searched ~my feelings~ Github plugins and found Dremel 3D20 who does basically what I was expecting. The dremmel plugin works very similar to this one. It has a custom file writer and a menu option to install the printer definition files.

I did similar thing and created a script to bundle the plugins so far into a **.curapackage** that contains both the work from @eskeyaar and mine. And it worked nicely! I dragged the package into a clean install of Cura and was able to install the resource files from the extension menu (Extensions -> Flashforge Finder -> Install files). After that, another restart is required (not sure if there is a way to refresh the printer definitions from the plugin call). Then I was able to add Finder as a printer and save .gx after slice (mime type worked!!!)

I am not sure this is the best way to do it. It would be very nice to have the finder printer definition, the profile and the GXWriter plugin bundled as part of the Cura install and not a plugin, but maybe the plugin is easier and faster to update.

eskeyaar, 2020-05-30:
> Small update. I searched ~my feelings~ Github plugins and found Dremel 3D20 who does basically what I was expecting. The dremmel plugin works very similar to this one. It has a custom file writer [...]

Hi,
Great to see its working nicely, am ready to give pull request of machine def to cura, but you have to submit plugin in the portal as @Ghostkeeper mentioned earlier. in that way, it will be bundled as part of cura install.
And about the .curapackage, im not sure how that can be integrated into it. 
Still figuring out how i can pack the .curaprofile along with machine defintions. Is the new profile prints well now?

ronoaldo, 2020-05-31:
@eskeyaar I'll submit the plugin to the portal, it is now working very nice.

The profile prints very well! I guess stringing is still a temperature issue with my material. I have to lower it down to 195-200 to reduce oozle and strings regardless of the slicer/settings I use.

One thing I noticed was that the profile overwrites the temperature from the material settings. From a clean Cura install I loaded up the plugins and added the printer, then loaded the profile, then loaded my PLA and PETG material files (.fdm_material). After that, I sliced and noticed that, when I filter "temp" it will overwrite the material temp (200) with the profile temp (220). Is it possible to let the temperature be used from the material settings?

Another smaller issue is that the seam position is visible and end up being quite noticeable. The default profiles position them as hidden and that is working very well. The "mini darth vader" test print I did ended up having the seam at a sub-optimal place.

As far as supports, I manually switch always to the Tree supports. They are by far the better ones for this printer, because they are closer to the ones Flashprint builds and they are much easier to remove.

Finally, I just would like to know if we can have the 4 "intents" that would be helpful to starters and are similar to what Flashprint "basic" settings provide:

* Low (Faster): 0.3mm; 80mm/s speed; 3 top/bottom layers; 2 shells
* Standard: 0.18mm; 60mm/s speed; 3 top/bottom layers; 2 shells
* High (Slower): 0.12mm; 70mm/s speed; 4 top/bottom layers; 3 shells
* Hyper: 0.08mm; 50mm/s speed; 6 top/bottom layers; 3 shells

Picked the values bellow from the Flashprint UI. The .curaprofile, if I understood correctly, is a "final evel" in the hyearchy, so I'm not sure it can be bundled. You may need to add "quality" profiles and "intent" profiles maybe?

ronoaldo, 2020-05-31:
Note: just managed to publish the plugins, they ware under review now :)

eskeyaar, 2020-05-31:
> @eskeyaar I'll submit the plugin to the portal, it is now working very nice.
>
> The profile prints very well! I guess stringing is still a temperature issue with my material. I have to lower it down [...]

Great!
I hope I can work out the "quality" Profiles with Low, Standard , high and hyper settings soon.

Regarding the seam, Can I see the picture and sliced gcode? so that i will try to make seam at the rear side and hidden. 

The Temperature in profile i set to was 220, but still if you change the temp, it should reflect the same. its not working?

ronoaldo, 2020-05-31:
> Great!
> I hope I can work out the "quality" Profiles with Low, Standard , high and hyper settings soon.
> 
> Regarding the seam, Can I see the picture and sliced gcode? so that i will try to make seam at the rear side and hidden.

Sure. Here is the set of pictures, and attached are the .gx generated by the printer. gcode is embedded at the line 3 and beyond of the file.

Pictures:
https://imgur.com/a/A3OEcSs

G-code files: 
[Cura-issue-7829-Gx_Files.zip](https://github.com/Ultimaker/Cura/files/4706997/Cura-issue-7829-Gx_Files.zip)

> The Temperature in profile i set to was 220, but still if you change the temp, it should reflect the same. its not working?

It does work, but it overwrites the value I have set at my custom material, so this kind of breaks the Material settings in that regards. I have to manually set the temp. and other settings for the material again at each print job.

ronoaldo, 2020-06-01:
Another day, another test and it still works like a charm! @eskeyaar the profile gives very good prints!

Did a bit of digging into Cura source here are some suggestions:

Maybe we could convert the .curaprofile into a quality profile like this: https://github.com/eskeyaar/Cura/blob/master/resources/quality/tronxy/tronxy_0.4_PLA_extra.inst.cfg

and it is linked with the extruder and definition files. I learned that we can just unzip the profile and it gives the text files for easy edit so we could extract them and place them as resources in your PR:

From the zip:

File: finder_santytesting
```
[general]
version = 4
name = SantyTesting
definition = fdmprinter

[metadata]
quality_type = draft
setting_version = 6
type = quality_changes

[values]
adhesion_type = skirt
layer_height = 0.18
layer_height_0 = 0.27
speed_slowdown_layers = 1
travel_retract_before_outer_wall = True
```

File: ff_finder_extruder_0_#2_santytesting
```
[general]
version = 4
name = SantyTesting
definition = fdmprinter

[metadata]
quality_type = draft
setting_version = 6
position = 0
type = quality_changes

[values]
bottom_layers = 3
cool_fan_full_layer = 1
cool_min_layer_time = 8
cool_min_speed = 15
default_material_print_temperature = 220
expand_skins_expand_distance = 0.1
infill_enable_travel_optimization = True
infill_pattern = zigzag
line_width = 0.48
material_final_print_temperature = 220
material_flow = 109
material_initial_print_temperature = 220
meshfix_maximum_resolution = 0.05
optimize_wall_printing_order = True
retract_at_layer_change = True
retraction_amount = 1.3
retraction_combing_max_distance = 10
retraction_speed = 30
skin_preshrink = 0.1
speed_print = 50
speed_travel = 100
speed_wall_x = 35
top_layers = 7
travel_avoid_supports = True
travel_compensate_overlapping_walls_enabled = False
z_seam_type = back
z_seam_x = 100
z_seam_y = 120
```
I would suggest taking out or changing the z_seam values to make it "hide the seam" and would remove the temperature settings so I believe they will inherit from the Material settings if the user has one. Not sure if just unpacking those into the resources/quality folder would do the trick but worth testing out!

eskeyaar, 2020-06-01:
Hi @ronoaldo , Thanks for heads up. 
I tried the method, but i can't able to see the settings after i put them in quality folder, its still shows cura default ones.
I also changed few things on the profile side and i spent nearly 2hours on this quality profile config :-( to add, but no luck!
Here is the settings i saved as inst.cfg,

[general]
version = 4
name = normal
definition = fdmprinter

[metadata]
setting_version = 15
type = quality
quality_type = Normal
global_quality = true

[values]
bottom_layers = 3
cool_fan_full_layer = 1
cool_min_layer_time = 8
cool_min_speed = 15
default_material_print_temperature = 210
expand_skins_expand_distance = 0.1
infill_enable_travel_optimization = True
infill_pattern = zigzag
line_width = 0.48
material_final_print_temperature = 210
material_flow = 109
material_initial_print_temperature = 210
meshfix_maximum_resolution = 0.05
optimize_wall_printing_order = True
retract_at_layer_change = True
retraction_amount = 1.3
retraction_combing_max_distance = 2
retraction_speed = 30
skin_preshrink = 0.1
speed_print = 50
speed_travel = 100
speed_wall_x = 35
support_bottom_enable = False
support_infill_rate = 10
support_interface_density = 50
support_interface_enable = True
support_interface_skip_height = 0.2
support_line_distance = 4.5
support_z_distance = 0.18
top_layers = 7
travel_avoid_supports = True
travel_compensate_overlapping_walls_enabled = False
wall_0_wipe_dist = 0.4
z_seam_relative = True
z_seam_type = back
z_seam_x = 100
z_seam_y = 120

ronoaldo, 2020-06-01:
@eskeyaar try changing the definition tag at general to the same ID o the finder def.json id:

```
[general]
version = 4
name = normal
definition = finder
```
Assuming this is what links the quality parameters with the printer in use, based on the way tronxy_x is done:

```
[general]
version = 4
name = Extra Fine Quality
definition = tronxy_x
```

ronoaldo, 2020-06-01:
Also, maybe removing the "global = true"; that looks like the reason it is not using my material temperature and only assuming the 220º one...

eskeyaar, 2020-06-01:
> @eskeyaar try changing the definition tag at general to the same ID o the finder def.json id:
>
> ```
> [general]
> version = 4
> name = normal
> definition = finder
> ```
>
> Assuming this is what links the [...]

oops, yes, i did tested with "finder" too, no result. Thats why i changed back to fdmprinter to just test, but no luck.
Regarding the temp? - even if you changes these value, it does always takes up 220?
[image]
I just changed value and checked gcode, it does change for me. 
[image]
See, it has 2 areas at temp 200, first layer temp and rest of layers. So please check again, and make sure u have changed all temp in settings.

ronoaldo, 2020-06-01:
Hi! I apreciate you effort into getting the quality definitions as part of this, this will greatly improve the overall experience.

About the quality profile not loading, I found something comparing the files for another printer that has quality profiles for several layer heights, monoprice mini v2. I sent you a [PR](https://github.com/eskeyaar/Flashforge-Finder-/pull/1) with the values but it is basically:

1/ Enable the attribute "has_machine_quality": true,
2/ Enable the attribute "prefered_quality_type": "standard",
3/ In the .inst.cfg file, specify version to 13, other values don't work for quality profiles.
4/ In the .inst.cfg file, specify the quality = "standard" to mach.

This worked with success when I tested, without the need to import a profile, just by providing the configurations:

[image]

The side effect is that it does not "inherit" any other ones, so it may be tricky to handle all. A simple change would be: copy and rename changing the values, say, for layer height, top-bottom layers and permieter shells; this would mimic the quality changes from Flashprint so they look reasonable changes to me.

About the temperature thing: sorry if I'm being imprecise here. I can set a different temperature, what I am suggesting is that the temperature setting should belong to the "Material", and unless the material overwrites the quality temperature, we should not set a default value for it in the profile. Temperature specificaly will vary with region, material brand, and others, so a sensible default would be the generic PLA one. If the quality does not shows any value, it will inherit the value from the current selected material. That is what I'm pointing out.. What I am saying is, if we specify the temperature in the quality settings, per-material temperature become useless. I was comparing the finder profile vs a "generic profile" with the sole aspect of temperature: assuming the PR will contain these quality changes, temperature would idealy be left out and be handled by the material settings for each vendor (generic, esun, user imported, etc.). Sorry if my interpretation is incorrect about how Cura merge the settings tree; it seems confuse on every place I read it!

Ghostkeeper, 2020-06-02:
You guys have been busy! I'll try to answer the things that I've missed during this extended weekend.

> Also, I added the "Pause at layer (Flashforge Fidner)" post-processing script to my repo... is this the correct way to deliver that one? Sadly, I tested all the available post-processing and this [...]

We won't accept a separate Pause at Layer script any more. In the upcoming Cura version all of them have been merged into 1 script. So if your printer needs modifications to pause, please add another option to the script in our current master branch.

> @Ghostkeeper how do I publish my .curaprofile along with machine definition?
Thanks

If you want to publish it in your plug-in you can do whatever you want. For instance you could call `cura.Settings.ContainerManager.importProfile(file_name)` to import a .curaprofile. For an experience consistent with other printers in Cura, you need to have them as quality profiles instead. A .curaprofile file is a custom user profile, which is a different profile type than the quality profiles. You can open up that file as a .zip and find the profiles inside for each extruder. It needs some metadata changes to know what material, nozzle, etc. it depends on then. Refer to the existing quality profiles in our repository.

> (not sure if there is a way to refresh the printer definitions from the plugin call)

No, there isn't. It reads the definitions on start-up only. You could add the definitions in the registration phase of the plug-in since that happens as one of the first things during start-up (before loading machines), but you can't add them during runtime.

> I am not sure this is the best way to do it. It would be very nice to have the finder printer definition, the profile and the GXWriter plugin bundled as part of the Cura install and not a plugin, but maybe the plugin is easier and faster to update.

I don't think Ultimaker will accept the GXWriter plug-in to be bundled, because that would mean that we need to maintain it as well. We don't have a way to maintain it since we don't have this FlashForge printer. And we also don't have any incentive to maintain it since it doesn't benefit Ultimaker printers at all, so I don't really have an argument for my superiors as to why I'm spending time on that you see. So the only options really are either a) add the profiles to Cura (which are easy for us to maintain) but have the .gx file format as a plug-in in the Marketplace, or b) bundle everything in the plug-in, which is the approach you seem to be taking now.

> And about the .curapackage, im not sure how that can be integrated into it.

I think you're confused here. A .curapackage is a bundle containing either a plug-in or a material profile. It's an alternative way to distribute plug-ins rather than via the Marketplace. For my plug-ins I provide both a .curapackage file in Github Releases as well as a download in the Marketplace. The Marketplace, behind the scenes, just downloads the .curapackage and installs that.

> Is it possible to let the temperature be used from the material settings?

In the profile file you could specify that it needs to have e.g.:

    material_print_temperature = =material_default_print_temperature - 20

This would give it an offset from the temperature provided by the material. The material's temperature is adjusted in the `material_default_print_temperature` setting, while the `material_print_temperature` setting is actually used for the print.

> 3/ In the .inst.cfg file, specify version to 13, other values don't work for quality profiles.

Note that this is specific to Cura 4.6. For newer versions you'd need to run a version upgrade on it. This is part of the difficulty of supplying these profiles through a plug-in rather than through Cura's bundled profiles. Perhaps it's easiest to just supply the correct profiles for each Cura version and copy the correct one in depending on the current Cura version. That does mean that you need to update your plug-in with every Cura version though. You can also try to programmatically run the version upgrades, or to only update the `setting_version` metadata to `CuraApplication.SettingVersion`. However if you only update the `setting_version` you'd be missing out on changes that we make to the settings, such as how we're halving the value of Maximum Deviation for the upcoming release of Cura to get the same result due to a bugfix.

> The side effect is that it does not "inherit" any other ones, so it may be tricky to handle all. A simple change would be: copy and rename changing the values, say, for layer height, top-bottom [...]

The listed profiles are all of the profiles that there is a printer-specific global quality profile for. Or if the printer doesn't have printer-specific profiles, the global quality profiles (which is that standard list of 0.06mm through 0.6mm layer height).

> Sorry if my interpretation is incorrect about how Cura merge the settings tree; it seems confuse on every place I read it!

I've recently written some more global documentation about how this works. It's a complicated system. See: [Container Stacks](https://github.com/Ultimaker/Cura/blob/master/docs/profiles/container_stacks.md) and [Getting a Setting Value](https://github.com/Ultimaker/Cura/blob/master/docs/profiles/getting_a_setting_value.md) for a complete reference of how settings are computed.

ronoaldo, 2020-06-02:
> We won't accept a separate Pause at Layer script any more. In the upcoming Cura version all of them have been merged into 1 script. So if your printer needs modifications to pause, please add another option to the script in our current master branch.

Makes sense! I saw that there is already an option that should work with my printer (M25 g-code). I'll test the plugin from master.

> No, there isn't. It reads the definitions on start-up only. You could add the definitions in the registration phase of the plug-in since that happens as one of the first things during start-up (before loading machines), but you can't add them during runtime.

Thanks. This option sounds good: at plugin I can automatically install the profiles instead of triggering at menu item then. I feel that if @eskeyaar provide the machine def.json, extruder and the quality profiles I have tested as a bundled one, this would be beneficial. I'm doing this plugin now just as a means for testing the files but providing them as part of Cura seems more reasonable.

> I don't think Ultimaker will accept the GXWriter plug-in to be bundled, because that would mean that we need to maintain it as well. We don't have a way to maintain it since we don't have this [...]

Makes sense. I'll keep the GXWriter as a plugin then. Can the def.json file for the printer have two mime-types, with regular g-code as a fallback? This would make a clean install of Cura work with the printer; even without the added metadata this plugins provides the printer is capable to understand plain g-code.

> I think you're confused here. A .curapackage is a bundle containing either a plug-in or a material profile. It's an alternative way to distribute plug-ins rather than via the Marketplace. For my [...]

Thanks for clearing this up. I was thinking .curaprofile could ship any resource but I understood my mistake it by trial and error! Good old printf debugging for the win!
 
> In the profile file you could specify that it needs to have e.g.:
>
> ```
> material_print_temperature = =material_default_print_temperature - 20
> ```
>
> This would give it an offset from the temperature [...]

This looks good!

> Note that this is specific to Cura 4.6. For newer versions you'd need to run a version upgrade on it. This is part of the difficulty of supplying these profiles through a plug-in rather than through [...]

This reinforce the idea of submitting the quality profiles as part of the PR then... I'm OK having this for newer versions of Cura only, and there is always the option to download the .curaprofile from the eskeyaar repo so this looks good to me.

> The listed profiles are all of the profiles that there is a printer-specific global quality profile for. Or if the printer doesn't have printer-specific profiles, the global quality profiles (which is that standard list of 0.06mm through 0.6mm layer height).

Makes sense. I think I was able to figure out by trial and error how to use the eskeyaar profile as quality profile, changing the metadata, and tested them. LGTM!

> I've recently written some more global documentation about how this works. It's a complicated system. See: [Container [...]

This is very useful thank you!

Ghostkeeper, 2020-06-03:
> Can the def.json file for the printer have two mime-types, with regular g-code as a fallback?

Yes, the `file_formats` metadata entry is a semicolon-separated fallback list. So you'd write something like "file_formats": "application/xgcode;text/x-gcode"` (but then using the correct MIME type for your .gx files). But you don't really need to use this because if there is no available writer, it'll use .gcode anyway which is always available.

ronoaldo, 2020-06-05:
I have uploaded the GXWriter package to the marketplace and also a plugin that allows users to use the definition files as well as quality profiles with Cura.

@eskeyaar can I help with anything else beyond the profile -> quality profile conversion? Kindly let me know if I can run any other print tests. I'm using them for some different jobs and results are looking good!

mahtDFR, 2020-06-05:
@ronoaldo I see your plugin is waiting in the portal, I'll get it through asap.

eskeyaar, 2020-06-05:
> I have uploaded the GXWriter package to the marketplace and also a plugin that allows users to use the definition files as well as quality profiles with Cura.
>
> @eskeyaar can I help with anything [...]

@ronoaldo Sorry being of busy over few days. Yes, any help in Profile > Quality profile to bundle with machine def. I will work few things today and post an update. 
Can i pm you?

ronoaldo, 2020-06-05:
@eskeyaar yes, feel free to PM me. I'm ronoaldo at Gmail too, if you prefer a chat.

I did the conversion and sent a PR to your repo, along with some test print results. I guess some other changes are needed to allow PETG and TPU filaments to work as well, probably just some small changes to the files. But I got the way the files need to be for them to be read as quality profiles by Cura.

ronoaldo, 2020-06-05:
> @ronoaldo I see your plugin is waiting in the portal, I'll get it through asap.

Thanks!

eskeyaar, 2020-06-09:
> Yes, the `file_formats` metadata entry is a semicolon-separated fallback list. So you'd write something like `"file_formats": "application/xgcode;text/x-gcode"` (but then using the correct MIME type [...]

I tried to implement this as @ronoaldo gave me as PR, I just tested the definition in cura 4.6, after slicing it tries to save as .3mf file, (gcode available in the drop-down list), without gx plugin. But with adding Gx writer into plugin directory, i get .gx as file option directly to save. 
Am doing wrong?

ronoaldo, 2020-06-09:
My PR was wrong actually, I typed the wrong separator and the wrong
mime-type. Kind test with ";text/x-gcode" at the end and it should fix it.

amitn, 2020-07-26:
Hi,

I am looking to add support for Flashforge Adventurer 3, and I was wondering how FlashPrint is doing the autodiscovery of printers in the network, with some Wireshark I'm now able to do auto discovery, and in the process of a proof of concept based on `UM3NetworkPrinting` plugin for FlashForge Adventurer 3.

Apparently the printer is listening for a multicast packet on `225.0.0.9:UDP:19000` and with a crafted payload `c0a8012b46510000` (I have not ideal what the payload is) it will reply with the printer name:
```
0000   4d 79 20 33 44 20 50 72 69 6e 74 65 72 00 00 00   My 3D Printer...
0010   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
0020   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
0030   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
0040   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
0050   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
0060   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
0070   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
0080   e1 00 00 09 22 c3 2b 71 00 08 00 00               ....".+q....
```

Below is the sample Python for sending the magic packet

```python
import socket
import time

def send():
    server = socket.socket(socket.AF_INET, socket.SOCK_DGRAM, socket.IPPROTO_UDP)
    server.settimeout(5)
    server.bind(("192.168.1.43", 18001))
    message = bytes.fromhex('c0a8012b46510000')
    while True:
        print("Sending: %s" % message)
        server.sendto(message, ("225.0.0.9", 19000))      
        data, addr = server.recvfrom(1024)
        print("received message: %s" % data)

if __name__ == "__main__":
    send()
```

Magic packet:
```
0000   01 00 5e 00 00 09 98 af 65 b1 3e 65 08 00 45 00   ..^.....e.>e..E.
0010   00 24 1b 2d 00 00 02 11 fa bf c0 a8 01 2b e1 00   .$.-.........+..
0020   00 09 46 51 4a 38 00 10 c4 42 c0 a8 01 2b 46 51   ..FQJ8...B...+FQ
0030   00 00                                             ..
```

Reply:
```
0000   98 af 65 b1 3e 65 32 4a 26 0a 86 5c 08 00 45 00   ..e.>e2J&..\..E.
0010   00 a8 7b fb 40 00 40 11 3a 9e c0 a8 01 30 c0 a8   ..{.@.@.:....0..
0020   01 2b 4a 38 46 51 00 94 69 37 4d 79 20 33 44 20   .+J8FQ..i7My 3D 
0030   50 72 69 6e 74 65 72 00 00 00 00 00 00 00 00 00   Printer.........
0040   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
0050   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
0060   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
0070   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
0080   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
0090   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
00a0   00 00 00 00 00 00 00 00 00 00 e1 00 00 09 22 c3   ..............".
00b0   2b 71 00 08 00 00                                 +q....

```

amitn, 2020-07-27:
After reviewing the "magic" payload, the pyload, is actually the SRC IP and the SRC UDP port for the reply with some padding:

The Magic: `c0a8012b46510000` Translate to:

- IPv4 `c0 a8 01 2b` --> `192.168.1.43`
- UDP PORT:`4651` --> `18001`
- Padding:0000

eskeyaar, 2020-07-27:
> Hi,
>
> I am looking to add support for Flashforge Adventurer 3, and I was wondering how FlashPrint is doing the autodiscovery of printers in the network, with some Wireshark I'm now able to do auto [...]

Hi,
Are you trying to connect via USB cable and print?

amitn, 2020-07-27:
> Hi,
> Are you trying to connect via USB cable and print?

@eskeyaar - I'm working on the  Flashforge Adventurer 3 WiFi auto discovery and in general networking printing to the Flashforge Adventurer 3 .

- The  .gx format is covered in this thread
- Sending .gx to the printer over wifi is covered at https://take4-blue.com/en/?cat=36 and in https://github.com/take4blue/Adventurer3Web

And I looking to see how the FlashPrint is doing the auto discovery of printers in the local network.

ronoaldo, 2020-08-03:
Hey all - long time no talk. I got issues with the printer, after a year of
use it was time to replace the nozzle. Got it sorted out this weekend and
I'm not fixing some build plate adhesion problems to further test the
profiles.

Perhaps it is worth split this in two PRs maybe? One for the printer
support (using the default profiles) and another for the custom profiles to
be imported? Having the printer and default cura profiles available can be
beneficial to users after all.

Amit, one question. I'm interested into sending the .gx file to the printer
as well, and I see you have some cool features implemented for Adventurer.
I could not find the part that do send the file to the printer but I
believe it is similar or maybe even compatible with Finder as well. Woud
you like to embed the monitoring features in Cura as well?

eskeyaar, 2020-08-12:
> 
> Hey all - long time no talk. I got issues with the printer, after a year of use it was time to replace the nozzle. Got it sorted out this weekend and I'm not fixing some build plate adhesion problems to further test the profiles. Perhaps it is worth split this in two PRs maybe? One for the printer support (using the default profiles) and another for the custom profiles to be imported? Having the printer and default cura profiles available can be beneficial to users after all. Amit, one question. I'm interested into sending the .gx file to the printer as well, and I see you have some cool features implemented for Adventurer. I could not find the part that do send the file to the printer but I believe it is similar or maybe even compatible with Finder as well. Woud you like to embed the monitoring features in Cura as well? Em seg., 27 de jul. de 2020 às 17:29, Amit Nishry <notifications@github.com> escreveu:
> […](#)
> -- Ronoaldo Pereira ronoaldo.com <http://www.ronoaldo.com/>

Hey Ronoaldo,
I thought this idea early itself. I will start making the printers def with default profiles for PR.

Thanks.

mletterle, 2020-10-06:
Just a quick note to thank everyone for their work on this, I just got a flashforge finder and being able to use Cura is awesome!

GClaser, 2021-01-15:
Hi All, I am new to creating printer profiles, how do I implement this profile with my flashforge finder?

Thank you

ronoaldo, 2021-01-17:
Hi! The profile can be imported from the repository or you can try to load the data from my extension repository. I guess my extension is not published at the marketplace yet (I probably missed some bits here and there).

ronoaldo, 2021-04-20:
Hi! I've published the plugin for the both the GXWriter and the Finder integration to Marketplace, but I have made some important changes to the 1.0.1 version (still under review). Is it possible to publish the new 1.0.1 version as well? Thanks!

With the plugins and with @eskeyaar PR this one can be closed!

closed 2021-06-21 by Ghostkeeper

Tharic99, 2021-07-15:
Should this now display in Cura 4.10.0? I don't see the FlashForge Finder in the list of printers in 4.10.0 and it wasn't in the release notes either.

Also, I assume this is for the Finder only, not the Adventurer 3. @amitn did you ever get the Adventurer 3 added?

Ghostkeeper, 2021-07-22:
> Should this now display in Cura 4.10.0? I don't see the FlashForge Finder in the list of printers in 4.10.0 and it wasn't in the release notes either.

No, that printer profile was not submitted as a built-in profile for Cura but is only supplied via the FlashForge Finder Printer Integration plug-in in the Marketplace.