Ultimaker/Cura #7830: [4.6.1] nozzle cools down to 60C after heating up and starting the print.

00ihsan, 2020-05-26:
**Application version**
4.6.1

**Platform**
Windows 10 x64

**Printer**
Creality3D Ender 3 Pro

**Reproduction steps**
changed the gcode to use variables so the bed and nozzle heat simulaniously.

**Screenshot(s)**
see gcode file.

**Actual results**
after heating up the bed and nozzle the bed stays warm and the nozzle drops to 60C.

**Expected results**
everything heats up and the print starts after the print completes everything cools down.

**Project file**
https://we.tl/t-APTZ9NSBmz

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4679909/cura.log)

**Additional information**
none

mahtDFR, 2020-05-26:
Check your start gcode, I think this might be the culprit:

``M140 S60 ; set bed temp

M104 S200 ; set extruder temp

M190 S60 ; wait for bed temp

M109 S60 ; wait for extruder temp``

You're telling the bed to heat up to 60,
then the extruder to  200
then the bed to 60 again
then the extruder to 60.

00ihsan, 2020-05-26:
Tnx

closed 2020-05-26 by 00ihsan