Ultimaker/Cura #7832: Extreme oozing and stringing in the default profile for Geeetech A10M

onfoot, 2020-05-26:
Hi!
I'm currently setting up my printer and from my initial testing 0.8mm retraction distance is far too low. After some tweaking, 7mm distance and 45mm/s velocity is loads better.

If somebody else confirms, I can make a PR for those changes.

https://github.com/Ultimaker/Cura/blob/e0734ff130ddaa53cc96e8fd7f602b9fb255db6d/resources/definitions/geeetech_A10M.def.json#L34

bastion-bastion, 2020-05-29:
HI
I have 6.8 mm retraction and 45 speed. Still have some issues with oozing.

Ghostkeeper, 2020-06-03:
Maybe it's got more to do with the printing temperature than the retraction settings then?

onfoot, 2020-06-03:
So far I've tried with even the minimal temperature values recommended for the filaments I have and there are still issues, even at 4mm and 35mm/s.

I'll try to ask around in some communities revolving around Geeetech's products.

closed 2020-06-19 by no-response[bot]