Ultimaker/Cura #7833: Missing image

leoheck, 2020-05-26:
Image here is missing
https://appimage.github.io/Cura/

64Florian, 2020-05-26:
For support team: 
[image]

The image link is 
https://github.com/Ultimaker/cura-build/packaging/screenshot.png
-> HTTP 404 Not found

nallath, 2020-05-27:
Where does it get the URL from? It should be https://github.com/Ultimaker/cura-build/blob/master/packaging/screenshot.png

leoheck, 2020-05-27:
The current and broken image address is this one
https://github.com/Ultimaker/cura-build/packaging/screenshot.png

nallath, 2020-05-28:
> The current and broken image address is this one
> https://github.com/Ultimaker/cura-build/packaging/screenshot.png

Yes, I know, but that URL can't ever work. But where does it get that (broken) url from?

Ghostkeeper, 2020-05-28:
That's here: https://github.com/Ultimaker/cura-build/blob/c43b8bce8c38e493bd28291402cc1f53b72e27b6/packaging/cura.appdata.xml#L15

Ghostkeeper, 2020-05-29:
I added a screenshot. The README of the main Cura repository badly needed one anyway.

closed 2020-05-29 by Ghostkeeper