Ultimaker/Cura #7834: Update Tool

leoheck, 2020-05-26:
Hi guys, I would like to use this tool to upgrade the AppImages
https://github.com/AppImage/AppImageUpdate/releases

[image]

It can't be used right now, because some info is missing. 
What do you think to support this? 
Thank you.

Ghostkeeper, 2020-05-27:
We have previously not been allowed to implement a self-updating system. The reason is that Ultimaker's marketing department wanted downloads to go through their website, lest people forget that Ultimaker's Ultimaker Cura™ by Ultimaker is made by Ultimaker...

Lately there has been some talk to change that, but I can't promise anything. But if we were to build anything like this, we'd build it for all platforms that we support.

leoheck, 2020-05-27:
@Ghostkeeper thanks for the info. The AppImage looks a promising thing for me here in Linux. But IMHO half of the features is not enough.

Maybe if the login inside the App does more instead of just showing Ultimaker printers and nothing else, it could be used to bound users with the Ultimaker Brand. For me, this login gives me just a blank/empty page, I can't use anything from there. The team's thing is useless. Even my printers/materials are not shared among my computers. I always have to set everything from scratch.

Ghostkeeper, 2020-05-28:
A lot of people use the back-up functionality as a way to transfer their settings and profiles to different computers.

leoheck, 2020-06-09:
Yes, I want this for the same purpose, but it looks that it is not working for me.
Also considering AppImage, I am not logged all the time I open Cura

yash-fn, 2021-06-11:
Possibly then create like a simple shell script utility that will display Ultimaker logo within terminal before going to ultimaker website and downloading the latest appimage file.

yash-fn, 2021-06-11:
Something to the tune of:
```
#!/bin/bash

#For Ultimakers Marketing Department
echo "Ultimaker's Ultimaker Cura™ by Ultimaker is made by Ultimaker"

LINK=https://ultimaker.com/software/ultimaker-cura

APPDIR=~/Applications

APPLINK=$(js -p 'decodeURIComponent(JSON.parse('"'$(grep -shoP '"https[^" ]*?\.AppImage"' <(curl -s $LINK))'"'));')

APPNAME=$(grep -shoP '[^/]*\.AppImage' <(echo $APPLINK) )

CURRENTAPPNAME=$(grep -E '[Cc]ura.*\.AppImage' <(ls $APPDIR))

if [ "$CURRENTAPPNAME" = "$APPNAME" ]; then 
	echo No Updates Available. Exiting.
else 
	echo 'Update Available. Upgrading '$CURRENTAPPNAME' to '$APPNAME;
	cd $APPDIR
	rm $CURRENTAPPNAME
	wget $APPLINK
	chmod +x $APPNAME
fi
```