Ultimaker/Cura #7835: 4.6.1 Controls via USB not correct and fails to heat hot end.

ndconline, 2020-05-27:
**Application version**
4.6.1

**Platform**
Windows 10

**Printer**
Ender 5 Pro (not listed in options so selected Ender 5)

Issue 1:

1. Connect printer via USB
2. Go to monitor section
3. Printer controls are reversed. Z axis for example when you click the up arrow, it moves the bed down. 

Issue 2:
When starting print from USB, the hot end settings are ignored so the printer will try to print with the hot end at room temperature. Bed heating does work though. 
Work around: If I click preheat for both bed and hot end and wait for them to both get to temperature before I select print from USB it will work.

Ghostkeeper, 2020-05-27:
Issue 1 is intended to "move the nozzle up", i.e. go to a higher Z coordinate. If your printer moves the bed down rather than moving the gantry up, that's fine. The result is correct nonetheless.

I can't reproduce the second issue ony UM2. The g-code sent via USB is the same as what is being saved to file. Is there anything wrong with the g-code when you save it to a file? Which g-code commands does your printer need to heat up?

koisdigala, 2021-01-17:
Issue 2 - I have this same problem, same workaround.  I'm not new to 3d printing, but am new to using Cura to send USB print job to an Ultimaker 2+.

Ultimaker 2+ with Cura 4.8.0 on MacOS 10.15.7
Used built-in filament profiles (Generic PLA) and tried a derived PLA filament profile too.

koisdigala, 2021-01-17:
Should close this bug as duplicate of 4927

fvrmr, 2021-01-22:
We will close this issue because it is a duplicate of #4927

closed 2021-01-22 by fvrmr