Ultimaker/Cura #7836: [4.6.1] machine Start/End G-code font unavailable, renders squares

mjrider, 2020-05-27:
**Application version**
(The version of the application this issue occurs with.)
4.6.1 

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Linux Mint and Linux Debian Buster
both have an intel videocard.
Almost all font packages are installed

**Printer**
(Which printer was selected in Cura?)
Ender 5

**Reproduction steps**
Open de machine settings

**Screenshot(s)**
[image]

**Actual results**
see picture

**Expected results**
readable text in de start/end gcode text areas

**Log file**
Could not find lines relating to font issues in the log, but this is on the stdout.
```
Fontconfig warning: line 5: unknown element "its:rules"
Fontconfig warning: line 6: unknown element "its:translateRule"
Fontconfig warning: line 9: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 8: unknown element "description"
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 76: non-double matrix element
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 76: non-double matrix element
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 84: saw unknown, expected number
Fontconfig warning: "/etc/fonts/conf.d/11-lcdfilter-default.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/11-lcdfilter-default.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/11-lcdfilter-default.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/30-metric-aliases.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/30-metric-aliases.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/30-metric-aliases.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/40-nonlatin.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/40-nonlatin.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/40-nonlatin.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/45-generic.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/45-generic.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/45-generic.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/45-latin.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/45-latin.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/45-latin.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/49-sansserif.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/49-sansserif.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/49-sansserif.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/50-user.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/50-user.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/50-user.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/51-local.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/51-local.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/51-local.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/60-generic.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/60-generic.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/60-generic.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/60-latin.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/60-latin.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/60-latin.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/65-fonts-persian.conf", line 34: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/65-fonts-persian.conf", line 35: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/65-nonlatin.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/65-nonlatin.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/65-nonlatin.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/69-unifont.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/69-unifont.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/70-no-bitmaps.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/70-no-bitmaps.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/70-no-bitmaps.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/80-delicious.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/80-delicious.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/90-synthetic.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/90-synthetic.conf", line 5: unknown element "its:translateRule"
```

**Additional information**
Tested with app images
4.4.1 doesn't have this issue
4.5.0 does have the issue

this is the only application with this issue, and only in this field. under materials the description field for example works fine

rburema, 2020-05-27:
Hm, the face that we haven't changed anything relevant near that code leads me to believe it might be a caching issue.

Could you please try to edit `~/.config/cura/4.6/cura.cfg`, and under the `[view]` header put `force_empty_shader_cache = True`? I'd like to know if that solves the problem (at least what Cura is concerned, there might still be something unrelated going on with your fonts).

mjrider, 2020-05-28:
Sorry, but that didn't help.

it's also strange, this is the only applicaties i have issues with, and the `Description` and `Adhesion Information` text area's don't have the problem. so it's not something generic i did try to do all possible steps in flushing font caches that didn't help either

seemsindie, 2020-06-13:
+1
[image]

smartavionics, 2020-06-13:
Those areas, along with the extruder start/stop code areas, are using a font called "fixed" from the theme. If you can provide a font called "fixed", it may use it? Hope this helps.

fieldOfView, 2020-06-14:
It is a bit more subtle than having a font named "fixed". The "fixed" font in Cura is defined here:
https://github.com/Ultimaker/Uranium/blob/e789e46ba29252c5333c6a615464abb492197232/UM/Qt/Bindings/Theme.py#L249
Uranium is asking the system font database to return the "standard" fixed-width font. Unfortunately it does not log what that fixed-width font is.

smartavionics, 2020-06-14:
FWIW, what I get at that location on Ubuntu 16.04.6 is monospace,9,-1,2,50,0,0,0,0,0

smartavionics, 2020-06-14:
Given that we appear to supply some noto fonts and that there is a noto fixed-width font, perhaps it would be a good idea to supply that font as well and explicitly ask for it rather than relying on the system to provide a fixed width font?

fieldOfView, 2020-06-14:
Originally the idea was that the dialog used the system default fixed width font, because the original Machine Settings dialog used "system styling" (like eg the Preferences dialog still does). With 4.0, part of the interface was changed over to use "Cura" theming, but other parts still use system theming.

mjrider, 2020-06-15:
brainstorming: could it be that the fixed font returned by the system doesn't support iso8859

if i run this command `xterm -fn "-*-fixed-*-*-*-*-*-*-*-*-*-*-*-*"`
or use xfontsel to select a fixed font, it returns a 2byte font
if i replace the selector with "-*-fixed-*-*-*-*-*-*-*-*-*-*-iso8859-*"

so this issue is visible because i have all available fonts installed instead of a very select set

Ghostkeeper, 2020-06-16:
Hmm, as far as I can see, Qt provides no way to filter that to select only an ISO8859 font though, if that would be a solution.

In the long term we have a mind to remove the system-styling from that dialogue. There are some significant hurdles to that though; the material manager has quite a lot of messy GUI code and bindings.

v3rm0n, 2020-06-17:
For me the issue is the PowerlineSymbols OTF font so for a quick fix I moved it to another location temporarily: `sudo mv /usr/share/fonts/OTF/PowerlineSymbols.otf ~ && fc-cache -fv`

mjrider, 2020-06-18:
i tested it by removing the powerline font package, and it indeed made the problem go away

so now the question is it cura or the font

aqshafei, 2020-06-19:
> i tested it by removing the powerline font package, and it indeed made the problem go away
> 
> so now the question is it cura or the font

I have testing this "hack/fix". it does works. However, I use powerline font with vim/neovim. so it is not a good fix for me since I could only temporary remove powerline, edit start and stop gcode. Then install powerline font back.

mjrider, 2020-06-20:
Some extra information
i unpacked the appimage, and replace the usr/bin/libfontconfig.so.1 with the one installed on my system. When i start cura with the new libfontconfig library it starts to work.

so maybe just updating the fontconfig library dependency in the build process is enough to fix this issue

tomrudnick, 2020-06-29:
I can confirm that if I replace libfontconfig.so.1 in the package the problem goes away using Debian 10 Buster with powerline-fonts installed.

rburema, 2020-06-30:
We think the problem is that the version of Linux we use in our build-system has become outdated. We where going to upgrade anyway because we want to support C++17 in the engine.

yankee14, 2020-08-07:
@mjrider @tomrudnick , I replaced the libfontconfig.so.1 with the one I found at /usr/lib/x86_64-linux-gnu/libfontconfig.so.1, and also tried symlinking it. I get `fontconfig.so.1: undefined symbol: FT_Done_MM_Var`. Is there something else I need to do?

tomrudnick, 2020-08-07:
@yankee14 Maybe your libfontconfig version is too old or just different. I've got Version 1.12.
Whats the output of the following command?

ls -la /usr/lib/x86_64-linux-gnu/ | grep 'libfontconfig'

My output looks like:
`-rw-r--r--   1 root root   500786 Nov  5  2018 libfontconfig.a
lrwxrwxrwx   1 root root       23 Nov  5  2018 libfontconfig.so -> libfontconfig.so.1.12.0
lrwxrwxrwx   1 root root       23 Nov  5  2018 libfontconfig.so.1 -> libfontconfig.so.1.12.0
-rw-r--r--   1 root root   281592 Nov  5  2018 libfontconfig.so.1.12.0
`

yankee14, 2020-08-07:
@tomrudnick thanks for fast reply!

```
$ ls -la /usr/lib/x86_64-linux-gnu/ | grep 'libfontconfig'
-rw-r--r--   1 root root   502258 May 15 05:55 libfontconfig.a
lrwxrwxrwx   1 root root       23 May 15 05:55 libfontconfig.so -> libfontconfig.so.1.12.0
lrwxrwxrwx   1 root root       23 May 15 05:55 libfontconfig.so.1 -> libfontconfig.so.1.12.0
-rw-r--r--   1 root root   281600 May 15 05:55 libfontconfig.so.1.12.0
```

tomrudnick, 2020-08-07:
@yankee14 Ok hm, as far as I know this symbol is defined in libfreetype. Check if the libfreetype6 package is installed.
And maybe check if the libfreetype.so files exists by: ls -la /usr/lib/x86_64-linux-gnu/ | grep 'libfreetype'

yankee14, 2020-08-07:
Yes, it seems I have that.

```
$ apt policy libfreetype6
libfreetype6:
  Installed: 2.10.2+dfsg-3
  Candidate: 2.10.2+dfsg-3
  Version table:
 *** 2.10.2+dfsg-3 1000
       1000 http://ftp.us.debian.org/debian testing/main amd64 Packages
         50 http://ftp.us.debian.org/debian unstable/main amd64 Packages
        100 /var/lib/dpkg/status
     2.9.1-3+deb10u1 750
        750 http://ftp.us.debian.org/debian stable/main amd64 Packages
```
```
$ ls -la /usr/lib/x86_64-linux-gnu/ | grep 'libfreetype'
-rw-r--r--   1 root root  1086098 Jul 15 07:10 libfreetype.a
lrwxrwxrwx   1 root root       21 Jul 15 07:10 libfreetype.so -> libfreetype.so.6.17.2
lrwxrwxrwx   1 root root       21 Jul 15 07:10 libfreetype.so.6 -> libfreetype.so.6.17.2
-rw-r--r--   1 root root   788392 Jul 15 07:10 libfreetype.so.6.17.2
```

tomrudnick, 2020-08-07:
@yankee14 Ok maybe it's because you got a newer version of the library. I've got version 2.9.1 on my Debian 10 System. I've read that the new version got some problems with Wine and other applications but I couldn't find any changes according to FT_Done_MM_Var.
FT_Done_MM_Var isn't a symbol, it's just a function that should free the memory. I don't have enough knowledge about freetype and can't reproduce it at the moment, but maybe you can downgrade freetype.

yankee14, 2020-09-01:
Still present in Cura 4.7

Ghostkeeper, 2020-09-03:
We've tried replacing our build server with a newer version of CentOS but had to pause that due to other DevOps tasks with greater priority (like the notarization of the MacOS build).

yankee14, 2020-09-12:
Still present in Cura 4.7.1

ika108, 2020-11-13:
Still present in Cura 4.8.0

anilcancakir, 2021-01-06:
I had the same issue. It should be about powerline fonts and AppImage font detection. I found a workaround. I'm using the configuration file for editing my g-code settings.

Cura is using `$HOME/.local/share/cura/$VERSION` folder to keep settings. If you open `definition_changes` folder in there, you can find your printer settings as a text file.

For example I added my custom start g-code by editing `Creality+CR-10S+Pro_settings.inst.cfg` file.

```[general]
version = 4
name = Creality CR-10S Pro_settings
definition = creality_cr10spro

[metadata]
type = definition_changes
setting_version = 16

[values]
machine_start_gcode = ; Script based on an original created by tjjfvi (https://github.com/tjjfvi)
; An up-to-date version of the tjjfvi's original script can be found
; here:  https://csi.t6.fyi/
; Note - This script will only work in Cura V4.2 and above!
; --- Global Settings
; layer_height = {layer_height}
; smooth_spiralized_contours = {smooth_spiralized_contours}
; magic_mesh_surface_mode = {magic_mesh_surface_mode}
; machine_extruder_count = {machine_extruder_count}
; --- Single Extruder Settings
; speed_z_hop = {speed_z_hop}
; retraction_amount = {retraction_amount}
; retraction_hop = {retraction_hop}
; retraction_hop_enabled = {retraction_hop_enabled}
; retraction_enable = {retraction_enable}
; retraction_speed = {retraction_speed}
; retraction_retract_speed = {retraction_retract_speed}
; retraction_prime_speed = {retraction_prime_speed}
; speed_travel = {speed_travel}
M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
	M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
	M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
	M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
	M220 S100 ;Reset Feedrate
	M221 S100 ;Reset Flowrate
	
	G28 ;Home
	
	G92 E0 ;Reset Extruder
	G1 Z2.0 F3000 ;Move Z Axis up
	G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
	G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
	G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
	G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
	G92 E0 ;Reset Extruder
	G1 Z2.0 F3000 ;Move Z Axis up
machine_width = 280

```

jonaslimads, 2021-04-18:
I can confirm it still is bugged in 4.8.0. The fix, as others have suggested, was:

```bash
# extract appimage
./Ultimaker_Cura-4.8.0.AppImage --appimage-extract

# replace Cura's packaged libfontconfig with system's own libfontconfig:
cp /usr/lib/x86_64-linux-gnu/libfontconfig.so.1.10.1 squashfs-root/usr/bin/libfontconfig.so.1
```

Then I've run `squashfs-root/AppRun` and it worked.

rburema, 2021-04-20:
This issue _might_ be resolved in 4.9(-beta). Explanation:

> We've tried replacing our build server with a newer version of CentOS but had to pause that due to other DevOps tasks with greater priority (like the notarization of the MacOS build).

We where forced to upgrade our Linux build-server to a newer version of CentOS for 4.9.x (since we needed to upgrade some of our supporting libraries, like Qt and Python). You could try the 4.9-BETA to see if this is still an issue (or see if the 4.9.0-final is already out which _should_ be along very shortly).

mjrider, 2021-04-20:
just checked, still present in 4.9

closed 2021-04-20 by mjrider