Ultimaker/Cura #7837: Import GCode (generated by Cura) as settings without creating a profile

BhaaLseN, 2020-05-27:
**Is your feature request related to a problem? Please describe.**
Most of my prints are single-use; ie. I load the model, pick a base profile, tweak some settings, save the GCode and then print it. Done. Never touching that Model or GCode again.

At least so I thought for most of them. But sometimes I _do_ have to revisit previously sliced models to address issues. Because I tend to rarely ever save the project files (which I only do when I know for a fact that I will be iterating in advance) I mostly have to make sure I don't mess with settings until I'm done - or re-import the previously sliced GCode as a custom profile (which is temporary, and I have to delete it afterwards to avoid having hundreds of them).

**Describe the solution you'd like**
Preferences > Profiles has an "Import" button that allows me to select a GCode file created by Cura, which will to a high degree of accuracy read the settings that are stored as comment at the end of a file.

I'd like to propose an additional menu option in the "Profile" dropdown under "Print Settings" (perhaps next to "Discard current changes"; see my suggestion below) which allows the same Import options as the Profile Management; _except_ that the settings only apply temporary (that is: they show the "Discard or keep changes" popup when the current selection is modified; then apply the base profile from settings and modify them as if the user set them by hand - leading to a modified profile **without** automatically saving them).

My use case is loading GCode files produced by Cura (and most likely: the same version; but at least the same major version) - but all options supported by the Profile Manager (such as `*.curaprofile` and `*.ini`) sounds like a good fit since the code to load them as profile is already there.

**Describe alternatives you've considered**
Using the Profile Manager and importing the file there is an option; but due to the temporary nature I have to go back and delete it afterwards; otherwise I end up with extra custom profiles that see very little than one or two uses. The mockup below still shows quite a few unused ones where I had to go back (such as "butterfly body 15deg supp" or "vase_simple_60mms"); but I haven't gone back to actually remove them yet due to being lazy.

**Affected users and/or printers**
Users that never save their projects (because they don't plan on getting back to them) sometimes have to reconstruct settings from already sliced files, and would benefit if the Cura UI could just re-apply settings from an existing file.

**Additional context**
Quick mockup suggestion on what it could look like:
[image]

rburema, 2020-05-27:
Those would be the settings received by the engine, as prepossessed by the formula's in the front-end. You'd loose a lot of connections between those, and it could be very hard to tweak some (or a lot) of them any more, depending on which settings are involved.

BhaaLseN, 2020-05-28:
Interresting, is this also the case when importing as custom profile? Because either way, I'd be fine I could skip the profile creation and _just_ apply them.

I mean, for the most part this isn't much of an issue for me, because I usually change values such as the wall count, top/bottom patterns or the infill settings; but rarely ever more specific ones (but when I do, I want them that way).

But then again, shouldn't the UI show the revert and calculate buttons anyways if this happens? It wouldn't if I create a profile (because thats the values the profile specifies); but it could/should when applying the base profile and then the changes recorded in the GCode comment.
Unless I'm missing some detail here that isn't obvious from just the GCode.

Ghostkeeper, 2020-05-29:
What do you think should happen if you already have unsaved changes when you import settings from a g-code file like this?

mahtDFR, 2020-05-29:
Thanks for the suggestion, it's a nice idea but we've decided that technically its difficult to implement, so we're going to focus on other priorities and we're not going to do this.

closed 2020-05-29 by Ghostkeeper

vkmaxx, 2025-09-01:
I made a free Cura Gcode inspector to check specific settings and reapply them before re-slicing models with newer Cura version, it's here:
https://www.maxxpreston.pl/gcode/