Ultimaker/Cura #7839: [4.6.1] "Select face to align to the build plate" doesn't quite align

fleutot, 2020-05-28:
**Application version**
Cura 4.6.1

**Platform**
OS: Debian GNU/Linux 10 (buster) x86_64 
CPU: Intel i7-7500U (4) @ 3.500GHz 
GPU: Intel HD Graphics 620 

**Printer**
Ultimaker 2+

**Reproduction steps**
"Select face to align to the build plate" doesn't seem to go all the way. I noticed that because the skirt on my print didn't go around the whole part, it went through the middle.

The model directly after opening:
[image]

After "Select face to align to the build plate", seen from below. The face is not completely red, only half of it seems to be touching the build plate:
[image]

After "Lay flat":
[image]

This is how I expect the downward face to look like after "Select face to align to the build plate".

fleutot, 2020-05-28:
This was a problem between the chair and the keyboard. The thin faces were not really parallel. Still, I don't get that "Lay flat" managed to places all the "not really parallel" faces flat on the build plate anyway.

closed 2020-05-28 by nallath

Ghostkeeper, 2020-05-29:
The lighter red and darker red happens because of the collision area shadow. That shadow is actually just a 3D plane. It's situated slightly above the build plate in order to prevent Z fighting. That's why faces that are on the build plate appear in a brighter red. This not only happens to faces that are 100% exactly aligned, but also to other faces as long as they fall beneath the 0.1mm Z coordinate or whatever that shadow is at.

The automatic "lay flat" option selects one of the lowest triangles to lay it flat on. It probably selected one that happens to make the rest of the triangles fall beneath the shadow. They are still not quite flat. Of course, the slicing also has a certain resolution due to the Initial Layer Height, so most likely the slice would succeed nonetheless.