Ultimaker/Cura #7840: "One part at a time" adds uneeded M190 lines, resulting in artifacts

truglodite, 2020-05-28:
**Application version**
4.6.1

**Platform**
Windows 10

**Printer**
A8

**Reproduction steps**
Use the 'print one at a time' print sequence options, and generate gcode with at least one part. No changes in bed temperature otherwise (same initial, print, and final bed temps). No other post processing gcode editors in use.

**Screenshot(s)**
N/A

**Actual results**
An excerpt from an example gcode (just the transition between parts):
```
;TIME_ELAPSED:144.485047
G1 F2400 E88.58544
G0 F7200 X116.138 Y193.786 Z5.5
G0 X95.753 Y98.544
M140 S47
M105
M190 S47
;LAYER_COUNT:2
;LAYER:0
M204 S1200
M205 X8 Y8
G0 X115.544 Y95.377 Z0.3
M204 S500
M205 X4 Y4
;TYPE:SKIRT
```

**Expected results**
Perhaps something along the lines of:
```
;TIME_ELAPSED:144.485047
G1 F2400 E88.58544
G0 F7200 X116.138 Y193.786 Z5.5
G0 X95.753 Y98.544
M140 S47
M105
;LAYER_COUNT:2
;LAYER:0
M204 S1200
M205 X8 Y8
G0 X115.544 Y95.377 Z0.3
M204 S500
M205 X4 Y4
;TYPE:SKIRT
```

**Project file**
N/A

**Log file**
N/A

**Additional information**
Having the M190 in the middle of the print causes a delay, which slows down the print and increases oozing artifacts (most printers will wait for the temp to 'stabilize' after any temp commands... often results in 2-3second pauses in my case). This may be acceptable in (rare?) use cases where each model may have different temps. The mid-print M190 is not needed if there is not temperature change commanded, but it is unfortunately added anyways and contributes to oozing artifacts. It would be better if the plugin would add M190 ONLY IF there is an actual change in temperature between parts/layers/etc.

Along similar lines (but a different issue), the mid-print M140's between each 'layer' are probably added from Cura processing it's 'initial and final' print temperature settings. Like the M190's, this to me seems like an improper implementation; it should know if the extra M140's are even required before adding them, rather than blindly adding them (at the risk of a print unecessarily pausing).

mahtDFR, 2020-06-03:
Thanks for the report, We discussed this morning and can't make this a priority, so it is deferred. We're open to pull requests fixing this.

visionok, 2020-09-18:
I'm having the same problem. Took me a while to figure out what was going on, when I finally did, it lead me here. Disappointed that it's not going to be a priority fix. But at least it's been noted as needing a fix so hopefully that will be forthcoming sometime in the future. In the meantime I'll probably use PrusaSlicer for printing "one at a time" since it doesn't have the same problem - but it does have another problem in that the preview does not show the order in which the parts will be printed, whereas Cura does show the print order.

visionok, 2023-06-11:
On 5.2.2 it seems to have been fixed. I tried a couple of examples and in both cases there was only 1 instance of "M190 S60" that placed near the very beginning of the gcode. Thanks.

MariMakes, 2023-06-13:
Hey @visionok.

Great to hear your issue has been resolved.
I'll close this ticket. 
Happy printing!

closed 2023-06-13 by MariMakes