Ultimaker/Cura #7841: Combining "Break up Support into Chunks" and "Support Wall Line Count" no longer working?

Silver-Streak, 2020-05-28:
**Application version**
4.6.1

**Platform**
Windows 10 64 Bit

**Printer**
Anycubic i3 Mega

**Reproduction steps**
1. Loaded the "cartridge_holder.stl" file into Cura from here:
2. Enabled "Break up support into chunks".
4. Set "Support Wall Line Count" to 1.
3. Sliced model.
4. Took screenshot.
5. Disabled setting.
6. Sliced model.
7. Took screenshot.
8. Enabled setting.
9. Set Wall Line count to 0
10. Sliced model.
11. Took Screenshot

**Screenshot(s)**
20mm "Chunks enabled", 1 support wall line
[image]
Inside of supports
[image]

2mm "Chunks enabled", 1 support wall line
[image]

No chunks enabled, 1 support wall line
[image]

20mm "Chunks enabled", 0 support wall line
[image]

**Actual results**
Wall lines are going around the entire support instead of the chunk, rendering this setting mostly useless. I've even attempted going down to a 2mm chunk size with no difference.

While my memory could be terrible, I remember that this used to work (even on this model) but no longer seems to function.

**Expected results**
Wall lines are added around _each_ chunk, keeping them as separate chunks but with their own walls.

**Project file**
https://we.tl/t-mJW9xfQV0I

**Log file**
[cura.log.txt](https://github.com/Ultimaker/Cura/files/4698383/cura.log.txt)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-06-02:
I don't think this ever worked. This is how it looks in Cura 3.4, when Support Wall Line Count got introduced:

[image]

Break Up Support In Chunks got introduced in Cura 3.0 and Support Wall Line Count in 3.4, so we certainly didn't take Support Wall Line Count into account when implementing that, since it didn't exist yet. The way that Break Up Support In Chunks works is by leaving out some of the "zags" of the zig-zag pattern. So it never really broke up the support area into multiple parts and thus wouldn't break up the surrounding wall line either.

Silver-Streak, 2020-06-02:
@Ghostkeeper Thanks for checking on this. I could have sworn it was treated differently before, but it appears I was incorrect.

Ghostkeeper, 2020-06-03:
We've discussed this, but decided that the implementation of making the extra support wall follow the broken up chunks is too complex. We've instead given the setting a warning if Break Up Support In Chunks is enabled. It's not a fix, but hopefully that makes it a bit more clear that the two don't combine well. I'll also mention it in the Settings Guide I think.

commit 6d1d6ca1 (2020-06-03): Add warning for Support Wall Line Count if breaking up support in chunks

It will be hard to break it up then.

Fixes #7841.
closed 2020-06-03 by Ghostkeeper