Ultimaker/Cura #7842: Add Octoprint @pause to Pause At Height script

agibson2, 2020-05-29:
**Describe the solution you'd like**
Octoprint has a feature where if it sees an @pause (lower case p) line then it sends it's own commands for pause and resume and also handles the actually resuming using the octoprint server's web GUI.  The current script rewrite has a repetier mode that sends @pause but it also sends other commands too.  Octoprint just needs the @pause put into the gcode and nothing else.

Any chance to get a new mode for pause at height to just put an @pause in the gcode and nothing else for Octoprint?

I maintain another script to do that but it would be great to have this integrated so that other users can take advantage of it without using a third party script.

I maintain a separate script here to do it...
https://community.ultimaker.com/topic/24105-post-processing-script-for-octoprints-pause-feature/

mahtDFR, 2020-05-29:
One for @fieldOfView I think

Ghostkeeper, 2020-06-02:
Doesn't the Repetier version work?

fieldOfView, 2020-06-02:
The repetier version also does funky head parking, as an inheritance from the original pause at height for repetier. The request for OctoPrint is to just have a single `@pause`. I am in dubio whether to make an OctoPrint flavor, or to make the parking optional for all flavors.

PS: @agibson2 the secret to entering `@pause` without it turning into a mention of the user named Pause is to surround it by backticks: `.

Ghostkeeper, 2020-06-04:
If making parking an option for all flavours is a bit of extra work, we can also do that later and then merge the OctoPrint pause with the Repetier pause. It wouldn't be very difficult to do.

agibson2, 2021-08-12:
Any chance of this getting implemented?  It would be really nice to have an integrated solution for Octoprint.

nallath, 2021-08-16:
Well, this isn't the kind of feature that will get attention from the main (eg; Paid by Ultimaker) developers. So the best chance is that someone makes a pull request. We would be more than happy to include it in that case.

fieldOfView, 2021-08-16:
Here's a PR: https://github.com/Ultimaker/Cura/pull/10288.
Disabling the head parking will make the repetier method suitable for OctoPrint. I have therefor also renamed the repetier method to repetier/octoprint.

I have not tested this PR extensively. Hopefully some of the posters in this thread can do that.

agibson2, 2021-08-16:
I would like to test it. Which files are changed? I would like to copy them to my C:\Program Files... to test it.  (I posted this to the PR on accident and deleted the comment there).

fieldOfView, 2021-08-16:
Only the `plugins/PostProcessingPlugin/scripts/PauseAtHeight.py` script was changed. See https://raw.githubusercontent.com/Ultimaker/Cura/e4a04c186f89855749c75bc864cb740ef2da67f3/plugins/PostProcessingPlugin/scripts/PauseAtHeight.py

agibson2, 2021-08-16:
When I dropped that into the folder,  The Pause At Height script disappeared from the list of scripts in the GUI on 4.10.0.  When I put the old one back it appears again.  Maybe something else needs to be done?  Here is the error that appears in the log...

`2021-08-16 13:00:09,497 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [204]: Begin loading of script: PauseAtHeight
2021-08-16 13:00:09,505 - ERROR - [MainThread] UM.Logger.logException [112]: Exception: Exception occurred while loading post processing plugin: Expecting value: line 62 column 38 (char 3083)
2021-08-16 13:00:09,509 - ERROR - [MainThread] UM.Logger.logException [116]: Traceback (most recent call last):
2021-08-16 13:00:09,513 - ERROR - [MainThread] UM.Logger.logException [116]:   File "C:\Program Files\Ultimaker Cura 4.10.0\plugins\PostProcessingPlugin\PostProcessingPlugin.py", line 206, in loadScripts
2021-08-16 13:00:09,516 - ERROR - [MainThread] UM.Logger.logException [116]:     setting_data = temp_object.getSettingData()
2021-08-16 13:00:09,519 - ERROR - [MainThread] UM.Logger.logException [116]:   File "C:\Program Files\Ultimaker Cura 4.10.0\plugins\PostProcessingPlugin\Script.py", line 92, in getSettingData
2021-08-16 13:00:09,523 - ERROR - [MainThread] UM.Logger.logException [116]:     setting_data = json.loads(setting_data_as_string, object_pairs_hook = collections.OrderedDict)
2021-08-16 13:00:09,527 - ERROR - [MainThread] UM.Logger.logException [116]:   File "json\__init__.py", line 370, in loads
2021-08-16 13:00:09,531 - ERROR - [MainThread] UM.Logger.logException [116]:   File "json\decoder.py", line 337, in decode
2021-08-16 13:00:09,535 - ERROR - [MainThread] UM.Logger.logException [116]:   File "json\decoder.py", line 355, in raw_decode
2021-08-16 13:00:09,539 - ERROR - [MainThread] UM.Logger.logException [116]: json.decoder.JSONDecodeError: Expecting value: line 62 column 38 (char 3083)
2021-08-16 13:00:09,541 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [204]: Begin loading of script: RetractContinue
2021-08-16 13:00:09,543 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [204]: Begin loading of script: SearchAndReplace
2021-08-16 13:00:09,551 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [204]: Begin loading of script: Stretch
2021-08-16 13:00:09,554 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [204]: Begin loading of script: TimeLapse
2021-08-16 13:00:09,556 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [204]: Begin loading of script: UsePreviousProbeMeasurements
2021-08-16 13:00:09,557 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [204]: Begin loading of script: PauseAtHeightOctoprint
2021-08-16 13:00:09,559 - WARNING - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadAllScripts [155]: Unable to create a folder for scripts: C:\Program Files\Ultimaker Cura 4.10.0\resources\scripts
2021-08-16 13:00:09,560 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin._restoreScriptInforFromMetadata [275]: Unknown post-processing script PauseAtHeight was encountered in this global stack.`

fieldOfView, 2021-08-16:
My bad, I made a typo.

Try https://raw.githubusercontent.com/Ultimaker/Cura/91b2a9b325701e065d4f7ff4497a47d595ffac1c/plugins/PostProcessingPlugin/scripts/PauseAtHeight.py

agibson2, 2021-08-16:
If I only had a penny for every typo I ever made :).

Test worked fine on Octoprint.  I put 2 pauses in place and they both paused where expected and the Octoprint gcode commands/config to move the nozzle and wait worked perfect.

Ghostkeeper, 2021-08-19:
That PR is merged. It didn't make it for Cura 4.11 since that is feature frozen, but it'll be in the release after that.

closed 2021-08-19 by Ghostkeeper

agibson2, 2021-08-22:
What is the intended behavior for the pause at height?  It appears that it pauses after the layer indicated and not before the layer indicated.  Is this expected?  I think part of the confusion is that the actual gcode starts at layer 0 but the GUI shows the first layer as layer 1.  So it pauses after the layer indicated in the GUI preview but before the layer number in the gcode which starts at 0.

After looking at the Cura preview and seeing that I wanted to pause before layer 9, I set the pause at height to pause at layer 9 and it paused after the layer 9 .  It seems to me that the meaning of 'at' would be before the layer is printed but maybe my interpretation was wrong.  It would be better to name the script pause before height or pause after height to be more clear or at maybe a note in the plugin screen to say which it would do.

Keep in mind that the first layer in gcode is layer 0 so layer 9 in gcode is actually layer 10.
------------------------------------
;added code by post processing
;script: PauseAtHeight.py
;current layer: 9
M83 ; switch to relative E values for any needed retraction
M84 E0
@pause now change filament and press ; Do the actual pause
-------------------------------------
[image]

Ghostkeeper, 2021-09-01:
Agibson2, your comment has nothing to do with Octoprint support for Pause at Height. It's better to post it in the appropriate issue report.

As far as I know it's supposed to pause at the start of the indicated layer number, as seen in layer view.

agibson2, 2021-09-02:
I want to note that I found out by accident today that Octoprint by default intercepts the Marlin M0/M1 commands and does not send the to the printer but instead does the same thing as seeing an @pause.  This is noted in their documentation that I was not aware of.  The big enhancement for this feature enhancement though is the 'Park Print' made optional so that only the M0/M1 or @pause is added and nothing else... letting Octoprint handle the gcodes to move the head and such.  So technically Marlin M0/M1 (with default Octoprint settings) would work the same as the Repetier/OctoPrint Method.

agibson2, 2021-09-02:
Well after testing Marlin M0/M1 today with 'Park Print' disabled with Octoprint. I definitely want to use the repetier/Octoprint method instead because the Marlin Method still adds temperature changes and such even with 'Park Print' disabled.  The Repetier/Octoprint method doesn't add anything except the @ pause which is what is wanted for Octoprint when you handle all the gcode in Octoprint for the print movement, etc.