Ultimaker/Cura #7847: [FQ] Solo mode

yet-another-average-joe, 2020-05-29:
**Is your feature request related to a problem? Please describe.**

Navigation in the the sidebar can be tedious when lots or all parameters are selected and a couple panels are opened

**Describe the solution you'd like**

A "solo mode", the way Adobe Lightroom manages panels : an option for displaying one panel at a time while automatically collapsing the other ones.

See here at 1:40 for a better description : https://www.youtube.com/watch?v=778MD_Q3TXM

**Describe alternatives you've considered**

Using Lightroom, but I didn't find the "Slice" button

**Affected users and/or printers**

everyone

**Additional context**

I think it could be a huge enhancement ; navigation between parameters would be much easier, reducing the need of the mouse wheel, and making parameters organization easier to learn for the beginer. It would also be usefull on small displays.

fieldOfView, 2020-05-30:
Perhaps you can add a screenshot of the problem; normally the panels are all the way on the left while the "sidebar" is all the way on the right. How would they interfere?

fieldOfView, 2020-05-30:
Ah, I see, you did not mean having toolpanels open (move, scale, per model settings), but you meant having multiple setting categories expanded. "Solo mode" would collapse all but one category.

yet-another-average-joe, 2020-05-30:
Yes : it's about the print settings categories.

Currently begining with Adobe Lightroom, I found this GUI option really smart.

Ghostkeeper, 2020-06-03:
I don't really see a problem with this. Good suggestion. I'll propose it to the team.

The name of the entry should probably be a bit more descriptive though.

Ghostkeeper, 2020-06-04:
We're putting this on our backlog.

fvrmr, 2021-06-03:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features from our community and from Ultimaker, we have decided to remove this feature request from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

fieldOfView, 2023-07-07:
You may be interested in this plugin:
https://github.com/fieldOfView/Cura-TabbedSettingsPlugin/

Also available in the Marketplace for Cura 5 and newer.

GregValiant, 2025-01-15:
FieldOfView's plugin works well.  I've had it installed for several Cura versions now.

@yet-another-average-joe I'll go ahead and close this.  Let me know of you feel that is premature.

closed 2025-01-15 by GregValiant