Ultimaker/Cura #7848: Cura 4.6.1 hangs on startup

sentrid, 2020-05-29:
**Application version**
(The version of the application this issue occurs with.)
4.6.1

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

OS Name	Microsoft Windows 10 Pro
Version	10.0.18362 Build 18362
OS Manufacturer	Microsoft Corporation
System Manufacturer	System manufacturer
System Model	System Product Name
System Type	x64-based PC
System SKU	ASUS_MB_CNL
Processor	Intel(R) Core(TM) i7-9700K CPU @ 3.60GHz, 3600 Mhz, 8 Core(s), 8 Logical Processor(s)
BIOS Version/Date	American Megatrends Inc. 1105, 6/6/2019
SMBIOS Version	3.2
Embedded Controller Version	255.255
BIOS Mode	UEFI
BaseBoard Manufacturer	ASUSTeK COMPUTER INC.
BaseBoard Product	ROG STRIX Z390-F GAMING
BaseBoard Version	Rev 1.xx
PCR7 Configuration	Binding Not Possible
Windows Directory	C:\WINDOWS
System Directory	C:\WINDOWS\system32
Boot Device	\Device\HarddiskVolume2
Locale	United States
Hardware Abstraction Layer	Version = "10.0.18362.752"
Time Zone	Eastern Daylight Time
Installed Physical Memory (RAM)	64.0 GB
Total Physical Memory	63.8 GB
Available Physical Memory	53.5 GB
Total Virtual Memory	73.3 GB
Available Virtual Memory	62.6 GB
Page File Space	9.50 GB
Page File	C:\pagefile.sys
Device Encryption Support	Reasons for failed automatic device encryption: TPM is not usable, PCR7 binding is not supported, Hardware Security Test Interface failed and device is not Modern Standby, Un-allowed DMA capable bus/device(s) detected, TPM is not usable
Hyper-V - VM Monitor Mode Extensions	Yes
Hyper-V - Second Level Address Translation Extensions	Yes
Hyper-V - Virtualization Enabled in Firmware	Yes
Hyper-V - Data Execution Protection	Yes

**Printer**
(Which printer was selected in Cura?)
Ender 3

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

Hangs on startup. 

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

Window launches, but only displays a 'blank white' window
[image]

**Actual results**
(What happens after the above steps have been followed.)
'Blank window'.  Waited 10+ minutes on 3 separate attempts

**Expected results**
(What should happen after the above steps have been followed.)
Would like to use Cura :)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)
N/A

[... 228 lines cut ...]
2020-05-29 18:42:38,938 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [112]: Initialized OpenGL subsystems.
2020-05-29 18:42:38,941 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [113]: OpenGL Version:  4.1.0 - Build 26.20.100.7262
2020-05-29 18:42:38,946 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [114]: OpenGL Vendor:   Intel
2020-05-29 18:42:38,948 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [115]: OpenGL Renderer: Intel(R) UHD Graphics 630
2020-05-29 18:42:38,949 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [116]: GLSL Version:    4.10.0
2020-05-29 18:42:38,951 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\default.shader]...
2020-05-29 18:42:38,967 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\selection.shader]...
2020-05-29 18:42:38,980 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\select_face.shader]...
2020-05-29 18:42:38,994 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\default.shader]...
2020-05-29 18:42:39,009 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\composite.shader]...
2020-05-29 18:42:39,029 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\overhang.shader]...
2020-05-29 18:42:39,049 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\striped.shader]...
2020-05-29 18:42:39,069 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\transparent_object.shader]...
2020-05-29 18:42:39,083 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\striped.shader]...
2020-05-29 18:42:39,103 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\xray.shader]...
2020-05-29 18:42:39,115 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\xray_composite.shader]...
2020-05-29 18:42:39,138 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\default.shader]...
2020-05-29 18:42:39,151 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\grid.shader]...
2020-05-29 18:42:39,166 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\platform.shader]...
2020-05-29 18:42:39,191 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\xray.shader]...

**Additional information**
(Extra information relevant to the issue.)
Have reinstalled a couple of times.  2 re-install attempts included full uninstall (including removing existing configuration files) as well as deleting the folders.
I can usually use Cura 1 time before it locks up on startup.

nallath, 2020-06-02:
The logs you provided are not from a full re-install. I don't know if it changes anything, but it is loading some stuff from a previous install

sentrid, 2020-06-03:
Thanks very much for the reply nallath.  For context, my 'uninstall' procedure was to run the uninstaller, then I deleted the "C:\Program Files\Ultimaker Cura 4.6" directory before I ran the installer again.

Are there other actions I should take here to clean out the detritus?  Is there another approach that I can use to find the detritus?

Thanks again.

nallath, 2020-06-03:
There are also some folders where it gets profiles / machines / plugins from. You can find them in the folder where the logs are stored (https://github.com/Ultimaker/Cura#logging-issues)

Could you try if making a zip backup of that folder and then starting it again helps?

sentrid, 2020-06-03:
That did the trick!! Thanks so much nallath.  I very much appreciate your help.

Ghostkeeper, 2020-06-03:
That still leaves us with the bug though. Do you still have that .zip? Maybe we can reproduce the bug with that.

sentrid, 2020-06-03:
You bet.  I made a zip of the entire directory.  

[cura.zip](https://github.com/Ultimaker/Cura/files/4723565/cura.zip)

Ghostkeeper, 2020-06-04:
Hmm, it's not hanging for me. I just get what are apparently your latest settings before it started freezing:
[image]

Maybe some Qt shader got corrupted. We've seen that before. But that wouldn't explain why it suddenly started working again when you reset the configuration folder.

Ghostkeeper, 2020-06-04:
In case you still wanted those settings you optimised, here is a zip file containing a project that should allow you to restore them (without reintroducing the freeze): 
[sentrids_settings.zip](https://github.com/Ultimaker/Cura/files/4728426/sentrids_settings.zip)

fieldOfView, 2020-06-04:
@sentrid do you have multiple displays? In that case you may want to *disable* the option to "Restore window position on start" on the General preferences page. On some systems, Cura will not start if it was previously closed when it was on a secondary display.

sentrid, 2020-06-04:
Thanks guys for the feedback.  That's good to know.  There aren't really any settings that I changed (much), so configuring a couple of settings wasn't a chore.

AndrewBryanScott, 2020-06-06:
i am having the same problem, cura ruins fine on my main display, but when i move it onto the second monitor it hangs, i have setup the default to not restore position so i can at least start the the application but really annoying how suddenly i cannot work on my second monitor. also cura 4.6.1

fieldOfView, 2020-06-06:
With that preference set, you can drag Cura to your secondary display and use it there.

AndrewBryanScott, 2020-06-08:
@fieldOfView that's what i have been trying but Cura now hangs once i have dragged it onto my secondary monitor, especially if i click the preferences menu option at the top of the screen. i am quite prepared to do a video of what is going on, maybe seeing it happen will be of use.
 (it is un-usable on a secondary screen for me now). it was working fine. it is also doing it on Cura 4.5. i can send logs but there not much there apeart from what has been reported. the last line of my logs are the same.

s\shaders\xray.shader

mahtDFR, 2020-06-08:
@AndrewBryanScott Logs and videos are always welcome, if we have more information we can debug more effectively and hopefully get you up and running faster

AndrewBryanScott, 2020-06-08:
[Cura4.6.1 Hang on Second Monitor.zip](https://github.com/Ultimaker/Cura/files/4744848/Cura4.6.1.Hang.on.Second.Monitor.zip)

please see video of hang on second monitor, and my log files from the video.

Also i must state that if the my settings are set to open Cura on the second monitor on start up i get exactly the same hang and it never starts, i had to disconnect my second monitor to get Cura to open on my primary screen in order to get it to the state where it opens there by default now.

mahtDFR, 2020-06-08:
Thanks for the detailed info, it really helps us understand what's happening.
There's nothing suspicious looking in your log as far as I can see. Are your drivers for the below up to date?

`4.1.0 NVIDIA 446.14
2020-06-08 09:44:01,748 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [114]: OpenGL Vendor:   NVIDIA Corporation
2020-06-08 09:44:01,751 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [115]: OpenGL Renderer: GeForce GTX 1060/PCIe/SSE2
2020-06-08 09:44:01,755 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [116]: GLSL Version:    4.0.0`

AndrewBryanScott, 2020-06-09:
I thought maybe drivers as well, so went and updated everything, problem still persists, i am about ready to download the code and run it in debug mode on mine to see if I can catch an error, i am a dev just don't normally work in Python and C and time is a very finite commodity for me at the moment.

Ghostkeeper, 2020-06-10:
Duplicate of https://github.com/Ultimaker/Cura/issues/6854.

closed 2020-06-10 by Ghostkeeper

Hookedon3D, 2020-06-29:
Ok, I have the same issue although, I have unchecked that option since I have a dual monitor setup but it still hangs and I can't even select preferences.

timkoers, 2020-07-02:
The issue is still here after a reinstall. Version 4.6.1, using a tripple monitor setup.

Below you can find a copy of my Roaming/cura folder.

[cura.zip](https://github.com/Ultimaker/Cura/files/4864655/cura.zip)

Hookedon3D, 2020-07-02:
> The issue is still here after a reinstall. Version 4.6.1, using a tripple monitor setup.
> 
> Below you can find a copy of my Roaming/cura folder.
> 
> [cura.zip](https://github.com/Ultimaker/Cura/files/4864655/cura.zip)
> 
> EDIT: It's fixed.
> The problem was caused by the GPU Tweak OSD overlay

I have an RX 580X where is the setting to change? TIA

Hookedon3D, 2020-07-02:
> The issue is still here after a reinstall. Version 4.6.1, using a tripple monitor setup.
> 
> Below you can find a copy of my Roaming/cura folder.
> 
> [cura.zip](https://github.com/Ultimaker/Cura/files/4864655/cura.zip)
> 
> EDIT: It's fixed.
> The problem was caused by the GPU Tweak OSD overlay

I have turned off "In Game Overlay" under settings/general and I'm using the latest drivers unfortunately  the issue persists.

achugb, 2020-07-13:
Autodesk Fusion360 demonstrates the same issue. Might be a bug with Qt according to some comments on their forums

achugb, 2020-07-29:
Submitted a bug post and process dumps to Autodesk and they came up with this. Can confirm that this works for both Cura and Fusion 360
https://knowledge.autodesk.com/support/fusion-360/troubleshooting/caas/sfdcarticles/sfdcarticles/Fusion-360-user-interface-menus-don-t-work-on-a-Windows-MSI-GT72S-Computer.html

Simrien, 2020-08-26:
I had this problem with version 4.6.2 after 24 hours.
Logged a ticket with support who said that they don't support Cura with 3D printers unless it's ultimaker.
Totally irrelevant as the printer isn't connected to my pc.

Uninstalled 4.6.2 and installed 4.61 which has been working fine for weeks up until last night.
Found this post tonight which talks about multiple monitors. 
Yes I have multiple monitors, no I can't disconnect one.
(It's attached through a KVM and cable management.)
The splash screen comes up on the right hand screen, Cura hangs on the left hand screen with a maximised white window.

So thought laterally and simply marked the left hand monitor as the main monitor in windows screen settings.
Bingo it worked and Cura fired up without hanging.
I de-maximised the window and moved it over to the right hand screen, closed it, marked the right hand screen as main and fired up Cura again without issue.

I had a look in the registry to find the setting mentioned in the thread above about restoring previous position.
There's very little in the registry under HKLM or HKCU so it's not configured there and couldn't find it in the Cura settings menu options.

Anyway, thanks for the tip and observation the hang is prevalent on multi-screen setups. :0)

achugb, 2020-08-29:
Have you checked the solution Autodesk came up with?