Ultimaker/Cura #7849: [4.6.1] incomplete preview, chunks of filament missing in layer view

bjjesteadt, 2020-05-30:
**Application version**
[4.6.1]

**Platform**
Windows 10 (10.0.18362, build 18362). Built-in GPU from Intel 8th Gen Core i5.

**Printer**
Creality Ender-3 Pro

**Reproduction steps**
Simply slice any model.

**Screenshot(s)**
PROBLEM, AS OCCURING IN 4.6.1:
[image]

WHAT SHOULD NORMALLY BE EXPECTED, AS IN 4.5.0:
[image]

**Actual results**
The print preview glitches out, missing portions of the filament that the slicer has created in the Gcode.  Note that the part still prints correctly, however the preview doesn't generate all of the skin lines, incorrectly allowing the internals of the print to be seen from the outside.

**Expected results**
All skin lines should be complete, and the preview should exactly represent what will actually be printed.

**Project file**
https://drive.google.com/file/d/10RSgqm41KT94_UKKfzLhu6S5XUfQiVpd/view?usp=sharing

**Log file**
dxdiag log file:
https://drive.google.com/file/d/1QGhL7wjDL9pBwkPwoqsi04Q-Vv2CWMIi/view?usp=sharing
cura log file:
https://drive.google.com/file/d/1tMa_qhBR264WW-d0B5rPV7mvODufT92d/view?usp=sharing

**Additional information**
This seems to occur no matter what; it does not matter what file I am slicing, what profile I have activated, or what settings I am using.  As stated, though, the models still _print_ correctly; they just display wrong in Cura.

nallath, 2020-06-02:
You enabled coasting. If you disable that it should show as normal.

Coasting basically replaces the extrusion at the last part of an extrusion path with a travel move. Since there is always a bit of oozing, material will still be printed. It's generally used to prevent stringing.

closed 2020-06-02 by nallath

bjjesteadt, 2020-06-04:
Crazy.  I don't know how I have never seen this before!  I (thought, at least) use coasting all the time.  Interesting.  
Thanks for the explanation!