Ultimaker/Cura #7850: Cura tries to override Stop Print command on printer while USB connected

rednosed, 2020-05-30:
**Application version**
4.6

**Platform**
Windows 10 on PC

**Printer**
Ender 5, Marlin 2.0.x

**Reproduction steps**
1. Start printing some object via USB from Cura (PC connected to printer via USB)
2. Stop printing via the printer (navigating through its menu) - in my case this made the printer head go to home position. After that, it returned to the print location (without adjusting Z height) and started extruding filament in the air above the print. Checking Cura it was still "printing" the object. I stopped the print via Cura software as well and the printer stopped extruding filament

fieldOfView, 2020-05-30:
If you start printing from Cura, you cannot stop printing from the printer interface. Cura does not know what you do on the printer interface, and just keeps sending gcode line by line.

nallath, 2020-06-02:
Closing this, as this is simply not something that Cura can do.

closed 2020-06-02 by nallath