Ultimaker/Cura #7851: [4.6.1] no 3rd-party printers listed

fergald, 2020-05-30:
**Application version**
4.6.1

**Platform**
Linux (tested on debian derivative and also Ubuntu 20.04)

**Printer**
None (I have an Ender 3pro but...)

**Reproduction steps**

- Download from https://creality.com/?do_action=action.download&DId=8
- Run ./Ultimaker_Cura-4.6.1.AppImage

**Screenshot(s)**

[image]

**Actual results**
Only ultimaker printers are listed

**Expected results**
Lots of other printers

**Project file**
None

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4705632/cura.log)

**Additional information**

With `Ultimaker_Cura-4.6.1.AppImage --appimage-mount` I can see that the files are there under
`/tmp/.mount_UltimalxWgfE/usr/bin/resources/definitions/creality_ender3pro.def.json`

fieldOfView, 2020-05-30:
There’s a scrollbar, but it is hidden until you use it. 

See https://github.com/Ultimaker/Cura/pull/7845

fieldOfView, 2020-05-30:
Duplicate of #7757, #7634, #7614

fergald, 2020-05-30:
Thanks. I don't have a mousewheel. Turns out of if I use cursor keys I can get down to other printers.

I had tried clicking the chevron to collapse the Ultimaker section but that's not clickable and tried searching with the name field. Glad to see that PR fixing all of these.

closed 2020-05-30 by fergald

fieldOfView, 2020-05-30:
Until the PR is accepted: you can also move your mouse over the area where the scrollbar would be, and it will magically appear.

Ghostkeeper, 2020-06-03:
> Thanks. I don't have a mousewheel. Turns out of if I use cursor keys I can get down to other printers.
> 
> I had tried clicking the chevron to collapse the Ultimaker section but that's not clickable and tried searching with the name field. Glad to see that PR fixing all of these.

You can also drag the scroll bar, after it appears. It behaves a bit weird though, a bit jumpy. That's another known issue.