Ultimaker/Cura #7852: Cura 4.6.1 freezes on reload of prepare tab when printing Lithophane

abs0lutek0ld, 2020-05-30:
**Application version**
4.6.0 & 4.6.1
**Platform**
AMD Ryzen 7 3800
32GB ram
NVIDIA 2070
Win 10 Pro

**Printer**
Ender 3 connected through octoprint

**Reproduction steps**
1. Used Cura's drag picture in to create Lithophane from photo because fun. 
2. Sliced and sent to print using octoprint plugin. (printed successfully)
3. About an hour later, attempted to go back to preview screen to start slicing another model for printing upon completion. 

**Screenshot(s)**

**Actual results**
Cura froze, screen went white.  Alt-F4 to kill.
On restart everything appeared to load but but when UI came up cura would freeze and white screen again.
**Expected results**
Returned to Prepare screen with lithophane on the build plate or blank build plate on restart.

**Project file**
[CE3PRO_LI_Test_v3.zip](https://github.com/Ultimaker/Cura/files/4705853/CE3PRO_LI_Test_v3.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4705858/cura.log)
[stderr.log](https://github.com/Ultimaker/Cura/files/4705861/stderr.log)

**Additional information**
A full wipe of and reinstall of cura fixes the problem. Problem can almost always be avoided if the project is cleared immediately after slicing and saving the G-code.  Other things were printed in this way (g-code staged in cura while waiting for another multi hour print to finish) and this problem was not encountered.

nallath, 2020-06-02:
I know there have been some issues in the past with video streams eating up a lot of memory. Does the problem also occur if you have no plugins installed?

Ghostkeeper, 2020-06-03:
I don't have an Octoprint printer. Does the problem only occur with Octoprint or is there something else that I can do that also reproduces the problem?

abs0lutek0ld, 2020-06-05:
It might be a memory leak or video issue.  I didn't pull up the resources tab and just closed the program so I can't say. I have re-installed my commonly used plugins and my customizations for bed size (direct drive cost me a few mm in z and x over a standard ender 3) so I can test to see if just slicing and leaving it for a few hours will work after I clear out a few backlogged prints. As I said If I'm quick enough (so far sub 5 min tested) it doesn't seem to hang but I haven't defined what quick enough means. As I've done this with other prints and octoprint I don't think that plugin is the problem but when I get it to hang again my only fix is a wipe and reinstall so I'll try again fresh with no mods.

closed 2020-06-24 by no-response[bot]