Ultimaker/Cura #7854: off set slicing.

bindedig, 2020-05-31:
**Application version**
4.6.1

**Platform**
Windows 10
**Printer**
CR10-s5

**Reproduction steps**
1. Sliced an item
2. send it to the printer like I always do via Octiprint

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
75% thru the print the y axis off set 5 mm failing the print.
resliced at 90 degree and x axis off set at 5mm failing the print on the same spot during transition from support to non support

**Expected results**
no offset

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fieldOfView, 2020-05-31:
This sounds like your printer skips steps. Try slicing slower, or specifically setting the travel speed lower.

bindedig, 2020-05-31:
All my speeds are at 40 mm. It never happened before 4.6.1. Can I download previous versions?

 

-Steve

fieldOfView, 2020-05-31:
See the “releases” here on github. Scroll down.

Ghostkeeper, 2020-06-03:
We have no way of reproducing this problem now. Can you point out the g-code that's causing the printer to offset by 5mm?

I'm currently also leaning to hardware failure. Try disabling Z hops or reducing the Z hop speed.

bindedig, 2020-06-03:
I have figured it out. :-)

When it upgraded to 4.6.1, I copied it to the old directory overwriting the
old files. So I uninstalled all of CURA and reinstalled 4.6.1

It now works fine, installing in older directories causes issues.

Thanks,

Steve

closed 2020-06-04 by nallath