Ultimaker/Cura #7855: [4.6.1] Hollow between lines if top / bottom pattern is set to lines

mondroidapk, 2020-06-01:
Hi,
When "the top / bottom pattern" is defined on "lines", there are spaces between each line but when it is defined on a concentric, the print is good. (see photo, left in line mode and right in concentric mode).
Best regards.
[image]

**Application version**
Cura 4.6.1

**Platform**
Windows 10

**Printer**
Anet A8

nallath, 2020-06-02:
We don't have access to an Anet printer, so it's a bit hard for us to test this. I have done a few prints over the weekend with the top bottom pattern set to lines on an Ultimaker 3, which did not have the issue that you seem to be having.

Are you sure you didn't change any other settings?

Ghostkeeper, 2020-06-03:
Is this visible in the g-code? Could you provide a project file for us, like the bug template asks?

mondroidapk, 2020-06-03:
Hi,
Attached is the test file I used.
I will receive a new printer in 10 to 15 days, I will try with it to see if the problem is with or not my Anet's firmware.
Best regards.

[test couche.zip](https://github.com/Ultimaker/Cura/files/4726331/test.couche.zip)

nallath, 2020-06-04:
The g-code seems to be correct.

Ghostkeeper, 2020-06-05:
Yeah, looks fine. I think this is a matter of tuning the printer or tuning the profile.

closed 2020-06-19 by no-response[bot]