Ultimaker/Cura #7856: [Feature Request] Custom buttons(Macros) on Monitor screen

pietchaki, 2020-06-01:
I like to use some gcodes to control my printer directly, I usually use Pronterface for that, as it have better gcode management (log screen with sent commands and their results) and it support custom buttons too, but I prefer using Cura with the "print via USB for convenience, otherwise would need to save gcode, load into pronterface and print from there.
The actions I put into buttons there are some leveling helpers (go to XY position, with a z-hop to avoid clamps), filament load/unload (heat the nozzle, wait for temperature, extrude or retract filament), set fan speed (as I have a single fan to cool the part and the extrusor, when I send a Pre-heat command, I also need to send a "M106 s128" to set fan speed to 50%, and also to set it to 0% after cooled off. Had too many problems with filament melting in the throat and clogging everything, so I set the temperature and fan speed from macros). Another example is my bed can be moved out of the printer enclosure, and in the "End Gcode" I already do that. Then I have a button to move the bed in and out for maintenance.
These are all very simple codes, I think the longest has 5 commands.
These macros also help other people use my printer without any trouble even if they don't know nothing about the code behind these actions. 

So, I suggest some way to create buttons to execute custom g-code macros from the Monitor screen.

Also, the ability to upload these new commands to (compatible, a.k.a. Ultimaker's) printer's LCD, can be useful to who use one. Not really for me tho.

Ghostkeeper, 2020-06-03:
Sorry, I'll have to defer this feature request. While the feature is nice to have, USB cable printing in general is not something we get to spend a lot of time on, since Ultimaker doesn't produce any printers that support that any more. We probably won't be able to implement this any time soon.

opok, 2021-12-17:
I would also love to have this feature. Sad that you don't want to add it.