Ultimaker/Cura #7861: [4.6.1] Tree support z distance not right

berberic2, 2020-06-02:
**Application version**
4.6.1

**Platform**
Linux

**Printer**
Monoprice Delta Mini

** Report**
If _**Tree Support**_ is selected (and normal support diabled) the distance between support and object is too big. As far as I can tell it’s add one layer to the normal distance calculated from _**Z Support Distance**_.
With _**Support Interface**_ enabled it gets even worse.

mahtDFR, 2020-06-03:
Please save a project file (File -> Save...), zip up the .3mf file and attach to this issue so I can try to reproduce it with the same settings as you.

mahtDFR, 2020-06-03:
Ah, looks like your right. I reproduced it in Windows 10 using the MP Mini Delta profile that ships with Cura.
Is that the profile you used?

[image]

Ghostkeeper, 2020-06-03:
Duplicate of https://github.com/Ultimaker/Cura/issues/5759. There are some more examples there and analysis of what's causing this.

closed 2020-06-03 by Ghostkeeper