Ultimaker/Cura #7865: Volumetric speed limits (auto speed)

AndrewEllis93, 2020-06-03:
**Is your feature request related to a problem? Please describe.**
When using adaptive layer heights, there is no capability to slow down the print so as not to exceed the volumetric limit of my hotend for thicker layers. This means I have to turn the print speed down for the entire print. 

I run a fast corexy printer so I almost always print near the volumetric limit of my hotend. I have to manually calculate the max speed for a given print using a formula (volumetric limit / layer height / line width) whenever I change my layer heights, line widths, or nozzle sizes. 

**Describe the solution you'd like**
An upper speed limit, in mm/3^s, either per material or per extruder. 

**Describe alternatives you've considered**
The workaround I have to use is slowing down the entire print (or using Prusa Slicer). 

**Affected users and/or printers**
I imagine this would be a bit more of a feature for advanced users. Perhaps printer profiles could include a default value for each printer to also benefit less experienced folks (i.e. preventing them from outrunning their hotend while experimenting with larger nozzles etc)

Ghostkeeper, 2020-06-04:
I think we've had this idea before as well, from one of Ultimaker's material engineers. I think it would be very useful for high speed printing profiles.

conceptual-perception, 2021-03-26:
I have a Prusa M3KS and ... changing out the nozzle from 0.4 to 0.8 ... broke Cura completely. I started a print and ... my Octoprint setup threw a M112 message. It took me a bit to understand what had happened, that Cura instructed the printer to send filament faster then the printer could safely extrude it. Thankfully, my printed detected this as a thermal runaway event and forced a reset.

I don't really see from a user perspective how this is possible, since the MVS value completely prohibits this. My E3D v6 hotend can only handle 15mm^3 a second. Where do I put this value into Cura? I thought adjusting the nozzle size would ... change other settings (it didn't), but ... idk if it's worth me going in and figuring out all the speeds for all the parts of the print. That seems like work software should do.

https://help.prusa3d.com/en/article/max-volumetric-speed_127176

```
The Maximum volumetric speed setting (MVS) is one of the most powerful features in PrusaSlicer. The MVS setting essentially creates a manager for the maximum amount of filament that the slicer will attempt to push through your 3D printer’s hotend.

This gives PrusaSlicer users a huge advantage. Rather than trying to calculate a range of speeds for combinations of different nozzle sizes, layer heights or extrusion widths, you can simply specify the speeds and settings you desire, then let MVS regulate speeds at slice time when and only if necessary.
```

I love Cura, but this missing, hurts.