Ultimaker/Cura #7867: [master] right click above rotate model arrows causes crash

smartavionics, 2020-06-03:
Just noticed that right clicking above one of the new rotate model 90 deg arrows reliably triggers a crash...

```
2020-06-03 11:41:12,530 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!
2020-06-03 11:41:12,531 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: Traceback (most recent call last):
2020-06-03 11:41:12,532 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "/opt/cura/lib/python3/dist-packages/UM/Qt/Bindings/MainWindow.py", line 190, in mouseReleaseEvent
2020-06-03 11:41:12,532 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:     self._mouse_device.handleEvent(event)
2020-06-03 11:41:12,532 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "/opt/cura/lib/python3/dist-packages/UM/Qt/QtMouseDevice.py", line 37, in handleEvent
2020-06-03 11:41:12,532 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:     self.event.emit(e)
2020-06-03 11:41:12,533 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "/opt/cura/lib/python3/dist-packages/UM/Signal.py", line 219, in emit
2020-06-03 11:41:12,533 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:     self.__performEmit(*args, **kwargs)
2020-06-03 11:41:12,533 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "/opt/cura/lib/python3/dist-packages/UM/Signal.py", line 332, in __performEmit
2020-06-03 11:41:12,533 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:     func(dest, *args, **kwargs)
2020-06-03 11:41:12,533 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "/opt/cura/lib/python3/dist-packages/UM/Controller.py", line 417, in event
2020-06-03 11:41:12,534 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:     self.contextMenuRequested.emit(event.x, event.y)
2020-06-03 11:41:12,534 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "/opt/cura/lib/python3/dist-packages/UM/Signal.py", line 219, in emit
2020-06-03 11:41:12,534 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:     self.__performEmit(*args, **kwargs)
2020-06-03 11:41:12,534 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "/opt/cura/lib/python3/dist-packages/UM/Signal.py", line 332, in __performEmit
2020-06-03 11:41:12,535 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:     func(dest, *args, **kwargs)
2020-06-03 11:41:12,535 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "/opt/cura/lib/python3/dist-packages/UM/Qt/Bindings/ControllerProxy.py", line 101, in _onContextMenuRequested
2020-06-03 11:41:12,535 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:     self.contextMenuRequested.emit(id)
2020-06-03 11:41:12,535 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: TypeError: ControllerProxy.contextMenuRequested[quint64].emit(): argument 1 has unexpected type 'str'
2020-06-03 11:41:13,316 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestError [361]: request[1994c1e2][get][PyQt5.QtCore.QUrl('http://example.com')][timeout=5][no-data] got an error 3, Host example.com not found
~/3dp/cura/Cura.git/build[mb-testing]$ 
```

One for @fieldOfView, I think.

Also, what's that HttpRequestManager error at the bottom about? example.com ? Is that related to sending the error report?

mahtDFR, 2020-06-03:
I could reproduce this too, it's up for discussion. Thanks for the report.

smartavionics, 2020-06-03:
Ah, just realised the example.com thing is spam generated by ConnectionStatus.

You're not seriously going to ping  the UM cloud every 10 seconds are you? For all Cura users?

closed 2020-06-08 by nallath