Ultimaker/Cura #7868: Material default temperatur is not used in printing

NilsRo, 2020-06-03:
**Application version**
4.6.1

**Platform**
Windows 10 X64

**Printer**
Anycubic I3 Mega S

**Reproduction steps**
1. Created a custom material as a copy from PLA temperatures as function
2. You can set in the custom profile to use the function to calculate the temperature

**Screenshot(s)**
[image]
[image]

**Actual results**
Printer started with 200 temperature if you use the default profiles as 200 is a fixed value there

**Expected results**
Change the default profile to calculate the temperature instead of using a fixed value.

**Project file**
-

**Log file**
-

**Additional information**
It is a profile setting only which is confusing for new users.

fieldOfView, 2020-06-04:
This is because the quality profiles for the Anycubic mega override the temperature that is set in the material:
https://github.com/Ultimaker/Cura/blob/master/resources/quality/anycubic_i3_mega/anycubic_i3_mega_normal.inst.cfg#L33

They also override other settings that probably should not be in quality profiles (material diameter, jerk and acceleration settings and others). But these profiles are not created or maintained by Ultimaker, so Ultimaker does not assume responsibility for them being correct.

Ghostkeeper, 2020-06-05:
For all we know, this profile was only designed to work at 200 degrees.

If you've got improvements to the profiles we can take a look to change the defaults, but we need to know exactly what to change then, preferably through a pull request. I suggest that you find some other users of that Anycubic printer, and come up with better profiles together.

closed 2020-06-05 by Ghostkeeper