Ultimaker/Cura #7877: Cura 4.6.1 does not generate tree supports

godlike64, 2020-06-05:
**Application version**
4.6.1, AppImage format

**Platform**
Gentoo Linux, Quadro M2200 Mobile

**Printer**
Ender 3 (I actually have an Artillery Genius which is not available, but I've been using the modified Ender 3 profile with great results for the rest of my prints).

**Reproduction steps**
1. Download either https://www.thingiverse.com/thing:3232756 or https://www.thingiverse.com/thing:3211969. Doesn't matter which, the problem seems to be global
2. Enable tree support in experimental, disable 'generate support'.
3. Slice the object.
4. Tree support settings does not seem to affect the end slice, it always comes out wrong, with no trees generated.
5. I tried to print a small piece (the first URL mentioned in 1.), so as to verify that it was not a display issue in Cura and trees would get printed. No trees were printed.

**Screenshot(s)**

**Actual results**
[image]
[image]
[image]
[image]

This is what printing the small piece yielded:

[image]
[image]
[image]

**Expected results**
This is what I would expect from a model like the Night Dragon one (not exactly this, as I did this in Meshmixer, based on the auto-generated supports, which didn't have great angles):
[image]

**Project file**
Either https://www.thingiverse.com/thing:3232756 or https://www.thingiverse.com/thing:3211969

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4733610/cura.log)

**Additional information**
While the screenshots show one set of settings, it does not appear to make any difference. I tried several changes to the settings and the end result is always the same.
I ended up printing the small piece so as not to waste too much material if it came out wrong.
I've also been browsing the older tree issues, and my problem appears to be nothing like that. In those issues some trees might pop up in the air. In my case there are no trees at all.
I don't know if I can do anything else from Cura once this issue happen to assist in troubleshooting it, so just let me know if there's anything else I could provide that would help.

Thanks,

martinsykes1, 2020-06-05:
Check your support wall count. I noticed that in earlier versions, wall count 0 would still generate tree supports but in 4.6.1 and later it had to be at least 1

godlike64, 2020-06-05:
Hmm, that looks to be it. I had previously set wall count to 0 because some supports are a pain to remove if they have an enclosing wall.

[image]

Ghostkeeper, 2020-06-08:
That's expected behaviour then.

The default for this setting is:

https://github.com/Ultimaker/Cura/blob/ac98caee5427bc73c558764c949f3d5a6d12c8b7/resources/definitions/fdmprinter.def.json#L4234

I.e. 1 wall for tree support, or some support patterns, and 0 for the more common support patterns.

However the Ender 3 profile overrides this to always be 1. Apparently that works better for those printers. Maybe their support needs more stability to be reliable.

closed 2020-06-08 by Ghostkeeper