Ultimaker/Cura #7879: where module is the models layout source code?

adairyao, 2020-06-05:
Dear everyone:
       I am a freshman and I am not familiar with the python language, I just want to know where the models layout source code, does any body can answer me?

mahtDFR, 2020-06-05:
Not sure what you mean exactly here. Do you mean the auto-arrange algorithm?
Images, pictures, or filling out the issue template will always help us understand and solve your issue faster. I've pasted it below for you to try again:

Please fill in the issue template so we are able to help you:

**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-06-08:
Here is the implementation of how Cura decides on a position for a model on the build plate: https://github.com/Ultimaker/Cura/blob/master/cura/Arranging/Arrange.py

Deciding on the position of multiple models at the same time is currently just a repeated iteration of that, positioning all models on the build plate one by one.

closed 2020-06-08 by Ghostkeeper

Ghostkeeper, 2020-06-08:
In broad strokes, the algorithm works like this:
* Create a 2mm pixel grid across the build plate.
* Fill in the pixels that collide with existing models on the build plate.
* Compute a bitmap for the new model you want to place.
* Try placing that bitmap on the pixel grid in the centre of the build plate.
* * If the new bitmap doesn't collide with any filled pixels, that's a correct location and the model is placed there.
* * Otherwise (if it does collide), try placing the model centred on a pixel adjacent to the previous centre and check again.
* It keeps checking different pixels as centre point, ordered by how close they are from the centre, until it either finds a free spot or all of the pixels are checked without finding a free spot.
* If there is no free spot, the model is placed next to the build plate, outside of the build volume.

adairyao, 2020-06-08:
Dear Ghostkeeper:
thank you very much! your answer is very helpful for me. my friends told me that Cura call the API of qhull-master(a c program library) to do the models layout algorithm, my friend is right? and where the cura project call these API? can you tell me? 
waiting for your favourable reply!

adairyao, 2020-06-08:
qhull-master c program library download address is "https://github.com/qhull/qhull"

Ghostkeeper, 2020-06-09:
We're calling upon QHull not directly but through Scipy, a library for scientific calculations. It's used in two places:

* https://github.com/Ultimaker/Uranium/blob/fa210f34c3db42720bcd083066c5fc39a7a00c1a/UM/Math/Polygon.py#L224 (Getting a 2D convex hull.)
* https://github.com/Ultimaker/Uranium/blob/fa210f34c3db42720bcd083066c5fc39a7a00c1a/UM/Mesh/MeshData.py#L509 (Getting a 3D convex hull.)

Note that this is not in Cura itself but in the framework called Uranium upon which Cura is built (intended for any 3D application with a backend to do calculations).

The 3D convex hull is used to create the collision areas underneath each model (which is used for that arrange algorithm). The shadow is 2D, but we're calculating a 3D convex hull because the 3D convex hull doesn't need to be recalculated when the user rotates the model. It can just rotate again. To get the collision area from that, we just project that hull to the build plate. The 2D convex hull is used to combine multiple objects if the user groups them.