Ultimaker/Cura #7882: Selective Horizontal Expansion

martinsykes1, 2020-06-05:
When slicing small D&D minis, some small features like swords need horizontal expansion because they are smaller than the nozzle diameter or layer height but I don't want to affect the whole model and lose detail in other areas.

I would like Cura to apply differential horizontal and vertical expansion to different parts of the model as necessary to make it printable without unnecessarily affecting areas that will print without it

I think this would benefit any user printing detailed models where appearance is more important than dimensional accuracy

For example, attached image shows a small diamond (e.g. a sword cross-section). The expansion would only affect the sharp edge which couldn't be printed and so is expanded to the nozzle size but the rest of the sword isn't affected.

[image]

Ghostkeeper, 2020-06-10:
This is to be implemented once this PR is merged, but it's a rather big change that we don't expect to be in the next release: https://github.com/Ultimaker/CuraEngine/pull/1210

Ghostkeeper, 2021-12-14:
This is fixed by https://github.com/Ultimaker/CuraEngine/pull/1210.

closed 2021-12-14 by Ghostkeeper