Ultimaker/Cura #7884: Will not open on MacOS

lcanino, 2020-06-05:
**Application version**
6.1

**Platform**
MacOS

**Printer**
no problems with my printer 
**Reproduction steps**
1. (Something you did.)
I updated my Macs software
2. (Something you did next.)
retried downloading  Cura
[image]

**Screenshot(s)**

[image]

**Actual results**
it bounces a few time and does not load even when I try to override and force it open

**Expected results**
it should open to the work bentch 

**Project file**
none

**Log file**
none that I know of

**Additional information**
No other issues that I am aware of

mahtDFR, 2020-06-09:
I guess you mean 4.6.1 here? Can you share a log please? This will help us work out why it's happening.

mahtDFR, 2020-06-09:
**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

fieldOfView, 2020-06-09:
The above dialog is shown before Cura has a chance to start logging.

mahtDFR, 2020-06-09:
Oh yeah. Actually this issue already exists on our backlog as CURA-7391
I suspect it's also a duplicate...

Ghostkeeper, 2020-06-10:
Duplicate of https://github.com/Ultimaker/Cura/issues/7389.

closed 2020-06-10 by Ghostkeeper