Ultimaker/Cura #7885: [4.3] Bug spotted when enabling Adaptive Layer Height and raising the model

mmahdim, 2020-06-05:
**Application version**
4.3.0

**Platform**
win10 64 bits, 

**Printer**
Tevo Tornado

**Reproduction steps**
1. raise the model (move in the z-axis), so that the model is floating in the air
2. enable supports
3. enable Adaptive Layer Height

**Screenshot(s)**
without Adaptive Layer Height
[image]
with Adaptive Layer Height
[image]

**Actual results**
Acts like the model file is empty and reports duration of print as 0 minutes.

**Expected results**
Should have sliced the file.

**Project file**
Any file

**Log file**
-

**Additional information**
-

nallath, 2020-06-08:
I've just tried this with the latest master and I could reproduce the issue.

mahtDFR, 2020-06-09:
This already exists on our backlog as CURA-6513
This ticket is a duplicate of #3183

closed 2020-06-09 by mahtDFR