Ultimaker/Cura #7886: Apply "top surface skin speed" to all walls on top surface

b4545, 2020-06-05:
**Is your feature request related to a problem? Please describe.**
My problem is that "extra skin wall count" prints faster than "inner wall speed" and way faster than "top surface skin speed", causing surface flaws.

**Describe the solution you'd like**
I would like some way to apply the top surface skin speed to the entire top surface.

**Describe alternatives you've considered**
I have had to turn off "extra skin wall count" to avoid the surface flaws.

**Affected users and/or printers**
Everyone

**Additional context**
Top layer speed with extra walls:
[image]
Prints much better without:
[image]

Ghostkeeper, 2020-06-10:
Currently the top surface skin settings only apply to skin on the top surface, not to other things like walls or infill. This also goes for other top surface skin settings, such as Top Surface Skin Pattern (which wouldn't make any sense for walls).

We could make it apply to other things as well (and rename it), but that would involve breaking up each of those features into separate areas as well. You'd get a border on your walls where it adjusts the speed. And it would make the outer wall on your lower layers also slower in those screw holes and chamfers, because the walls are exposed to the top.

As a workaround for now, I suggest reducing your wall count to 1 and increasing the Extra Skin Wall Count and Extra Infill Wall Count to 2. It will create more material where your skin touches infill, but with a flat model like yours that won't occur very often.

b4545, 2020-06-12:
Thanks for checking out my issue and replying. Here's what I thought:

> We could make it apply to other things as well (and rename it), but that would involve breaking up each of those features into separate areas as well.

I know the list of options grows and grows, and I understand that sometimes you have to resist adding every single option that pops up. Maybe this option doesn't quite make the cut.

> You'd get a border on your walls where it adjusts the speed. And it would make the outer wall on your lower layers also slower in those screw holes and chamfers, because the walls are exposed to the top.

The real issue is that if I want to have really fast inner wall speed in the middle layers, I have to accept that fast speed on the surface, which is not ideal. I'm fine keeping the outer wall speed the same for all layers. Additionally, "extra skin wall count" walls are printed **on the surface** at "top/bottom speed", so if you want extra skin walls you have to slow down your top bottom speed too, again not ideal.

> As a workaround for now, I suggest reducing your wall count to 1 and increasing the Extra Skin Wall Count and Extra Infill Wall Count to 2. It will create more material where your skin touches infill, but with a flat model like yours that won't occur very often.

I did plug this in and it would not help. I print with a relatively high top/bottom speed, so with extra skin walls the surface would print way too fast. I can't use extra skin walls for this reason. I am also forced to use a slow inner wall speed, because I want the inner walls on the top surface to print slowly. I haven't used extra infill walls before, so that's pretty interesting and I plan on playing around with it.

To summarize, think about adding a "top surface inner wall speed" and maybe outer also, but I understand you have to watch the number of options. Also, I think that you should change the extra skin walls **on the surface** to use the top surface speed instead of the top/bottom speed. That's it, thanks for your consideration.

mahtDFR, 2020-06-12:
We discussed this and feel we can't give it a priority at the moment, so I've had to defer it.
If you're working with flat surfaces here you could implement this with a post processing script as a workaround.

b4545, 2020-06-12:
Completely fair. Great work guys, I really love using your app.

yet-another-average-joe, 2023-04-17:
I came here after I searched how to reduce top inner walls speeds (and BTW discovered there's no suche option)

This works if the top is flat : add an Eraser + Modify settings for overlaps. The tricky thing is gettting the blocker to exactly intersect the top layer using the mouse.

Before :

[image]

After :

[image]

closed 2024-05-01 by github-actions[bot]