Ultimaker/Cura #7888: Cura 4.6.1 steals my Serial Port!

StubMandrel, 2020-06-06:
Cura 4.6.1

W10

Prus i3

1. Pulled out all my hair because Arduino was intermittently unable to access a USB-Serial com port
2. Discovered another application was using the com port
3. Confirmed it was Cura, as shutting it down releases the com port
4. Got really annoyed as there are no com port settings in Cura.

**Screenshot(s)**
N/A

**Actual results**
If I exit Cura I can then access my com ports

**Expected results**
Cura should not occupy com ports like some invading army crying out "All your Com Port are mine!"

**Project file**
N/A

**Log file**
N/A

**Additional information**
It's pretty obvious, CUra grabs any com ports it see in case a 3D printer is connected to one. This makes it quick and easy for non-techies to connect a printer that uses a USB-serial connection.

But it's lazy, sloppy coding that screws up any other apps that want to use a com port.  Cura should search available com ports only when a user connects a printer.

smartavionics, 2020-06-06:
Open the Cura marketplace and go to the installed tab and disable the USB printing thing. After that Cura won't dick with your serial ports.

dxgldotorg, 2020-06-07:
I had a problem with my wireless router crashing as I had left a USB-serial debug port attached.  It was pretty much open Cura, lose Internet and more importantly LAN.  Found the same fix and it stopped the router from crashing due to garbage on the serial port.

StubMandrel, 2020-06-07:
That didn't work for me... but has after a computer restart, which shouldn't have been needed but who cares, it's leaving me alone now.

I've a vague memory it interfered with my AVRISP too, but i haven't used that for nearly a year.

mahtDFR, 2020-06-09:
Maybe you guys can share some logs that I can take to the devs to try and solve this? We need some data as a starting point.

Ghostkeeper, 2020-06-10:
Duplicate of https://github.com/Ultimaker/Cura/issues/3449.

closed 2020-06-10 by Ghostkeeper

AndreKR, 2024-10-17:
I can't find the USB printing thing in the installed plugins. What exactly is it called?