Ultimaker/Cura #7892: Cura 4.6.1 Profilemanager values mismatch to current set values

AlbatrossXL, 2020-06-07:
I discovered that the CURA 4.6.1 profil manager is storing different values to the given values in advance.

Process / Steps:
I set i

1. n the slicing options values for density, speed, bed temperature etc.
2. I selected "create profil" to store this values in a new profile
3. I opened the profil manager to compare the values (!! NO changes inbetween ..!!)
4. I saw, that Cura took over wrong values into the profil manager (see screenshot)
5. The values are on several slicer options wrong. (see screenshot 1)
6. I repeated the process by chaning only one value (Bed Temperature from 50° -> 40°)
7. I repeated the check -> Now NO value has been stored in the ProfiManage ! (see screenshot 2)

**Application version**
CURA 4.6.1 

**Platform**
Win 10; NVIDEA GPU

**Printer**
Anycubic i3 mega

**Reproduction steps**
see above Step 6 and step 7

**Screenshot(s)**
2 Screenshots showing the mismatch between values in the menu and vlues in the profil manage

**Actual results**
-- 
**Expected results**
Values in the profil manger are equal to the cura-option menu 
OR at least
values in the profil manage in column "actual

" are equal to cura-option menu

**Project file**
n.a

**Log file**
n.a

**Additional information**
GCODE does also not reflect the values in the Slicing Menu;
Decreasing of BedTemperature is not in the GCode. (M140 command only for 50°C)

[image]
[image]

64Florian, 2020-06-07:
Hello,
Could you have a look in your "extruder 1" panel please ?

And is it possible for you to click on "Help" -> "show configuration folder" -> Your explorer should open a "4.6" folder, then zip this folder and add it in this ticket please 

Thank you

AlbatrossXL, 2020-06-09:
Hello,

please find attached the informations & values:

 >EXTRUDER 1: this panel is completly empty. No values are in
 >ZIP.-File is attached.

kind regards,

64Florian, 2020-06-09:
Hello @AlbatrossXL 
Unfortunately your attachment wasn't received in your ticket
Could you try to drag and drop your files in the ticket instead
https://github.com/Ultimaker/Cura/issues/7892
(_reply by email seem not properly works_)

Thank you

AlbatrossXL, 2020-06-16:
[4.6_Cura_conf.zip](https://github.com/Ultimaker/Cura/files/4788334/4.6_Cura_conf.zip)

closed 2020-06-16 by AlbatrossXL

64Florian, 2020-06-16:
Thank you @AlbatrossXL 
You have closed this ticket, is it solved for you ? 

@Ghostkeeper, It could be interesting to have a look in this case ...
I noticed AlbatrossXL use a profile named "Anycubic i3 Mega #2_user" and as I can see, many files use %23 for # encoding ...
I was able to see the same as AlbatrossXL show in his 1st screenshot on my IMac 

For me, this issue look like my ticket "_profiles settings not matching up in the application #7626_"
(_in my case I use "µ" char in my file name -> %C2%B5_)

Ghostkeeper, 2020-06-18:
I don't think this has anything to do with the `unquot_plus` encoding of the profile ID. The real problem here is that AlbatrossXL is comparing global settings as shown in the dialogue with the extruder settings shown in the normal settings list.

That makes it indeed very similar to #7626. This is both a UX problem (that there is a global tab at all) and a programming problem (that the extruder tabs are empty). The programming problem has been held back by the fact that this piece of code is apparently a bit complex. It's on our backlog though. The UX problem is a conflict. We'd like this panel to give a bit more detail or insight in how the profile is actually stored. It should only list the overrides (since here the profiles can be viewed independently of what lies underneath) and it should list them for all extruders. Under water we store the global settings separately, so it makes sense to also display them that way.