Ultimaker/Cura #7893: Spiral Vase mode start level

Cecchellone, 2020-06-07:
Hello I was suggesting a way to slice the lower part of an object in "normal" mode, and the top part in "spiral vase" mode.
It could be very usefull if the interface could specify a "start height" for the Vase Mode, so it will normally slice the bottom part, and a certain level it will start with the spiral vase mode.
_Obviously if needed it also needs to make a shell not to leave the infill below visible._

mahtDFR, 2020-06-09:
What's the use case for this? Can this not already be done using a cutting mesh?

fieldOfView, 2020-06-09:
> Can this not already be done using a cutting mesh?

No. Vase mode is not available as a per model setting, so neither can it be used with a cutting mesh.

Cecchellone, 2020-06-09:
The only way i found to do this, is to split the object in two parts, slice the _lower_ one with normal settings, slice the _higher_ one in spiral vase mode (but also removing the bottom shell and the adesion settings) and then putting them together via a Python script that **lift up the _top_ gcode's Z** commands of the _bottom_ gcode's max Z position.

It works but it obviously act strangely in matter of cooling, speed and temperature regulations.

mahtDFR, 2020-06-09:
Can you explain more about the application you have in mind here? Is it for practical purposes or is it just for aesthetics? I think this will have an influence on whether it's picked up or not.

Cecchellone, 2020-06-09:
**It’s kind of both**, I think it’s faster if I explain what I was trying to do.

I wanted to print a cylindric “bar equalizer” made of stacked led strips rings.
The model I made is made of two parts, an inner one, where the strips sticks to, and the outer one, that has to be made in spiral vase mode to let the leds visible from the outside.
Obviously the two parts must stick together, and for that purpose, both parts have a thread, the inner one acts as a screw, and the outer one as a bolt.

But to do this, the outer part needs its bottom section to be printed in “normal” mode, and the above section in spiral mode to let the light visible.
I hope I explained myself.

If needed i can provide the 3D models.

mahtDFR, 2020-06-10:
Thanks for the request, we discussed this in our stand up and have decided to defer it in favour of other priorities. We're open to pull requests to implement this.

Ghostkeeper, 2020-06-10:
This is a bit difficult due to the Smooth Spiralized Contours setting (which most people associate with spiralize), because it has special casing for the first and last layers. It tries to print the first layer ramping up and flattens the last layer.

Generally I'd love something like Simplify3D where you can specify a process for certain layers. However we've also seen that sometimes this is a bit more limiting than it needs to be. We'd rather go for an easier way to change settings in a certain volume.

clfaye, 2021-02-18:
I would also like to add this feature.  It was possible in Cura 2.5 (and I still use Cura 2.5 for this purpose).  I build lighting sculpture and I need a strong conventionally printed base and a 1 layer thick top.

clfaye, 2021-02-19:
More accurately, a "vase start level" feature wasn't available in Cura 2.5.  However, in Cura 2.5 you were allowed to "spiralize outer contour" as a per model setting.  So it was easy to have a solid model on the bottom and put a spiralized model directly on top.  After Cura 2.5 this is no longer permitted.  To get around this I either use Cura 2.5 (which is not ideal because it is so old) or I slice everything separately and then merge the gcode files making appropriate edits so that everything is seamless.  It is very time consuming.

clfaye, 2021-02-19:
Hey - we can do this ourselves!  If you go into the Cura program folder, resources, definitions and find "fdmprinter.def.json" - you can edit that file.  Find "Spiralize Outer Contour" and change "settable_per_mesh" to true.  Also change "smooth_spiralized_contours" "settable_per_mesh" to true.  Then both of these will show up as "per model settings" the next time you open Cura.  I haven't done a print with it yet, but I have sliced a model with a solid base and a spiralized top and it works just as it did in Cura 2.5.

clfaye, 2021-02-19:
Nope - doesn't work as expected.  It looks like it has spiralized in the slicer, but it really just creates a single walled print.  It doesn't smoothly rise the way that it is supposed to.  It creates a series of single layers so that the seam is clearly visible.  You can find the seam and the travel lines in the slicer preview.  So, I'm back to stitching gCode files together.

Ghostkeeper, 2021-02-25:
Looking at the code, the walls generation looks into the mesh settings for the spiralize setting (so it makes sense that you get only one wall and 5 in the base), but auxiliary routines look at the global settings for the spiralize setting, making this a global-only setting without modifications to CuraEngine. A couple of things that request the setting in a global context are:

* Determining which part gets spiralised and which doesn't (if you have multiple parts, only one actually gets a spiral path with gradually increasing Z).
* Polygon order optimisation with Surface Mode.
* Not retracting in layer change when spiralizing.

There are also a number of places where we request the setting from the global settings even though per-mesh settings would be available:

* Making sure the nozzle moves inside away from the wall when it's about to travel to a different part.
* Generating the line segments for a wall with correct width and speed.
* Generating the initial bottom layers.
* Skipping infill and skin generation if spiralizing.

And for some things we do actually request it from the per-object settings:
* Generating the correct number of insets.
* Determining which extruders get used (since for most layers Spiralize causes only the outer wall to be used).
* Determining the top surface skin areas.

So it seems we were quite consistently requesting it from the global settings, since Spiralize is a global setting. But for about half of those cases we could easily make it a per-object setting. Still, those first 3 cases are the reason why we can't make Spiralize a per-object setting. A solution is probably possible but it'd need some re-thinking to make it per-object.

Cecchellone, 2021-02-26:
I didn't really deep dived in the code, so maybe what I'm going to say it's meaningless, but it's worth a shot saying...
Could it maybe be simpler to set it up as a per-object setting **only if the printing sequence is "one at a time"?**
I don't even know if it's a conceivable option.

Ghostkeeper, 2021-02-26:
If the print sequence is one at a time, you can still print multiple objects at the same time by grouping them, so it's never really going to be a fix for routines that govern things in between models (like the ordering of polygons, or choosing which part gets spiralised). And you can even just have one model that splits into two pieces, like a Y shape standing upright.

ThorsenRune, 2021-06-10:
I have a usecase for this. It's like a outer cover for a handle in TPU where the first 5 mm needs to be a solid

shengyan, 2021-07-31:
I may just got a solution.
1. Turn on "Spiralize Outer Contour".
2. Set "Bottom Layers" to the height of the bottom section.

This is a Planaria trap cap I printed using this technic.
[image]
[image]

ThorsenRune, 2021-08-18:
I think it would be good if we could put a modifier so at a certain height it switches from/to spiral to normal mode. I would find it usefull to add a solid top to a spiral mode.

robertpdx, 2023-06-14:
In case anyone stumbles across this issue, I was not able to get the method suggested @shengyan to work, however it does work in Prusa 2.6.0-rc1.  I merged two models and sliced in spiralized mode, setting the bottom layers to match the height of the ring/holder of a lamp shade.  He's a Cura 5.2.2 preview and a  50% scale Prusa preview and Ender 3 Pro print.  I understand that these two slicers are sharing a lot of code now.  It would be a great feature to see in Cura.

Cura 5.2.2
[image]

Prusa 2.6.0-rc1
[image]

Print at 50% scale on an Ender 3 Pro:
[image]

B-Man1, 2023-08-11:
I would love to be able to cap off my vase mode prints

B-Man1, 2024-08-11:
This would be a great feature to have and seems easy to implement

ThorsenRune, 2024-08-12:
I definitely think that vase mode for a height interval should be available. Preferably by a modifier

Cecchellone, 2024-08-13:
Hi, I'm the one who proposed this feature in the first place.
I still think this could be a nice addition to Cura.

xX-MrN0b0dy-Xx, 2024-08-13:
Just joined the thread, i'd be interested to see this feature implemented too

PieTempesti98, 2024-08-13:
I think implementing this feature can be really useful and not so impossible to implement